PSP-Archive/ARK-4 #317: camera_patch_lite won't work

Phoebe-Galaxy, 2024-02-18:
### PSP/Vita Model

PCH-2000 (Vita Slim)

### ARK Version

FW 6.61 ARK 4.20.68 r8 ePSP (Standalone)

### Homebrew/GAME (if applicable)

UCUS-98653

### Plugins ( if applicable )

ULUS10041, ms0:/seplugins/gta_remastered_v2.prx, on

ULUS10160, ms0:/seplugins/gta_remastered_v2.prx, on

UCUS98653, ms0:/seplugins/camera_patch_lite.prx, on

### Steps to reproduce

1- Install ARK using Faster Ark
2- Install the plugin ArkRightAnalog
3- Using both the controller EBOOT from ArkrRightAnalog and the gta_remastered_v2 plugin the right analog works fine
4- Using the plugin camera_patch_lite on the other hand, won't work (in this case, trying to use it in God of War Chains of Olympus)

### USB Charge

Disabled

### Overclock

Always

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Always

### Inferno Cache

LRU

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

Off

### Hide DLC

Off

### Turn off LEDs

Disabled

### Verification of form.

- [X] I agree that I properly filled out the items listed above.

JoseAaronLopezGarcia, 2024-02-19:
At first glance it appears to be a conflict with ARK's built in controller patches (used for stuff like No Analog and key combo). I'll give it a try when I can.

JoseAaronLopezGarcia, 2024-02-19:
We found out that the plugin has a check in its code to only load in Adrenaline (this was because the plugin was made when only Adrenaline had second analog functionality).

This ARK pre-release should disable the check to let the plugin work:
https://github.com/PSP-Archive/ARK-4/releases/tag/v1708346216

Let me know if this solves the issue.

Phoebe-Galaxy, 2024-02-19:
Just tried the pre-release with both God of War Chains of Olympus and Spider-Man 2, the plugin works perfectly now.

Thank you!

closed 2024-02-19 by JoseAaronLopezGarcia