PSP-Archive/ARK-4 #318: Japanese and Korean characters are not displayed.

gilgilgan, 2024-02-20:
### PSP/Vita Model

PCH-2000 (Vita Slim)

### ARK Version

4.20.68 Rev 9

### Plugins ( if applicable )

When I change the language setting in settings to Japanese or Korean, no characters are displayed in the menu. 
In File Explorer, filenames in Korean or Japanese also don't display correctly.

### Steps to reproduce

change the language setting in settings to Japanese or Korean

### USB Charge

Disabled

### Overclock

Disabled

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Disabled

### Inferno Cache

Off

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

Off

### Hide DLC

Off

### Turn off LEDs

Disabled

### Verification of form.

- [X] I agree that I properly filled out the items listed above.

JoseAaronLopezGarcia, 2024-02-23:
It's a strange issue. It doesn't want to load those fonts from flash0, but the regular (latin) fonts load correctly.
If I make it so that the fonts are loaded from ms0 instead then it works fine.
We're currently looking for a solution.

Aneemasta, 2024-12-24:
FW 6.61 ARK 4.20.69 r151 cIPL: Currently Japanese and Korean localization of `CL` seems like working.
[image]
[image]

---
The other issue described in the OP is still there though. And it's not just the Japanese or Korean symbols can't be displayed properly by `File Browser`, Russian ones affected as well, regardless of selected `CL` localization:
[image]
[image]
[image]
[image]

JoseAaronLopezGarcia, 2025-03-16:
This was fixed already.