PSP-Archive/ARK-4 #321: By pass the anti privacy check in Flowers and Flowers: Natsu-Hen

NagaseKouichi, 2024-03-09:
### New Feature

Japanese Visual Novel Flowers and Flowers: Natsu-Hen have a anti privacy check. If you run the game from ISO file, when the game booted, it will ask you to install the game to the memory stick, after the installing it will ask you to pull out the memory stick, if you pull out the memory stick and insert it again, it will give you a error screen says check failed and wont boot game further.

You can use a SAVEDATA to bypass the check on PPSSPP, but that don't work on PSP. On PSP the game will give a system save data error.

Is there any way to  pass the anti privacy check?

krazynez, 2024-03-09:
Does it force you to install game data? Kinda poinless doing that when it already resides on the memory stick.

NagaseKouichi, 2024-03-09:
> Does it force you to install game data? Kinda poinless doing that when it already resides on the memory stick.

Yes, you must install the game or it will not go further.

krazynez, 2024-03-09:
> > Does it force you to install game data? Kinda poinless doing that when it already resides on the memory stick.
> 
> Yes, you must install the game or it will not go further.

Just found this: https://github.com/yurisizuku/psp-mschange

NagaseKouichi, 2024-03-09:
> > > Does it force you to install game data? Kinda poinless doing that when it already resides on the memory stick.
> > 
> > 
> > Yes, you must install the game or it will not go further.
> 
> Just found this: https://github.com/yurisizuku/psp-mschange might be worth a shot

Thank you.
I just tryed this plugin, but it doesn't work.
When the game ask me to remove the memory stick, I press select+down to emulate removing memory stick. Then the memory stick reading led blinks sometimes, so I think maybe it is emulating removing ms. But the game doesn't have any response, and stop at the screen ask me to remove the memory stick.

Maybe this plugin is not compatible with ARK-4?

krazynez, 2024-03-09:
> Thank you. I just tryed this plugin, but it doesn't work. When the game ask me to remove the memory stick, I press select+down to emulate removing memory stick. Then the memory stick reading led [...]

it does work just have to do it a few times it seems. Had to translate every dialog, it also checks your battery as well if its low or not.

[image]

krazynez, 2024-03-09:
> > Does it force you to install game data? Kinda poinless doing that when it already resides on the memory stick.
> 
> Yes, you must install the game or it will not go further.

Yeah I see that.

JoseAaronLopezGarcia, 2024-03-09:
This pre-release fixes it: https://github.com/PSP-Archive/ARK-4/releases/tag/v1710012849

closed 2024-03-09 by JoseAaronLopezGarcia

krazynez, 2024-03-09:
https://github.com/PSP-Archive/ARK-4/releases/download/v1710012849/ARK-4-v1710012849.zip

Tested and working @JoseAaronLopezGarcia fixed it so the plugin will work.

NagaseKouichi, 2024-03-10:
> This pre-release fixes it: https://github.com/PSP-Archive/ARK-4/releases/tag/v1710012849

> https://github.com/PSP-Archive/ARK-4/releases/download/v1710012849/ARK-4-v1710012849.zip
> 
> Tested and working @JoseAaronLopezGarcia fixed it so the plugin will work.

Wow, that is quick! I tryed this release and it's working perfectly, thank you!