PSP-Archive/ARK-4 #322: Data load error when load PSP game with custom launcher

NagaseKouichi, 2024-03-14:
### PSP/Vita Model

PSP-3000 (Slim & Bright)

### ARK Version

4.20.68 r11

### Homebrew/GAME (if applicable)

Trick x Logic: Season 1

### Plugins ( if applicable )

npdrm_free

### Steps to reproduce

1. load custom launcher
2. load Trick x Logic: Season 1 game
3. load savedata

### USB Charge

Always

### Overclock

Disabled

### Power Save

XMB

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Always

### Inferno Cache

LRU

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

On

### Hide DLC

Off

### Turn off LEDs

Disabled

### Verification of form.

- [X] I agree that I properly filled out the items listed above.

NagaseKouichi, 2024-03-14:
If I run "Trick x Logic: Season 1" from ARK-4 custom launcher, and load the savedata, it will give different result from if I load the game from XMB.

This is when I load the game from XMB, and what it should be:

[image]

And the charpters I have unlocked load without problem:

[image]

If I run the game from custom launcher, and then load the exact same savedata, I get this:

[image]

The charpters i already unlocked is in gray and can't be load (instead some charpters that I didn't unlock appeared). And if I load the charpters in gray, the game gives a error says "There is no story data, please insert the UMD or media with the data and run again"

[image]

NagaseKouichi, 2024-03-17:
Uploaded the savefile incase that you want to test it.
[NPJG00025SAVE00.zip](https://github.com/PSP-Archive/ARK-4/files/14625854/NPJG00025SAVE00.zip)

krazynez, 2024-03-17:
> Uploaded the savefile incase that you want to test it. [NPJG00025SAVE00.zip](https://github.com/PSP-Archive/ARK-4/files/14625854/NPJG00025SAVE00.zip)

Thanks I was testing this out the other day with no replication. That is helpful for sure. Ment to ask you the other day kinda got caught up with things.

NagaseKouichi, 2024-03-17:
> Thanks I was testing this out the other day with no replication. That is helpful for sure. Ment to ask you the other day kinda got caught up with things.

And I'm using this pre-release that fixes mschange plugin: https://github.com/PSP-Archive/ARK-4/releases/tag/v1710012849
Don't know if it has difference with the R11 version though.

krazynez, 2024-03-18:
Okay so just did this and still can replicate with your savedata. Both CL and XMB restult in same data that is accessible.

NagaseKouichi, 2024-03-18:
> Okay so just did this and still can replicate with your savedata. Both CL and XMB restult in same data that is accessible.

Do you test it with PSP or PSVita? And do you test it with v1710012849 pre-release version?

I find that there is no problem with ePSP on PSVita and I,m running 4.20.68 r11 release on my PSVita's ePSP.

I still have this problem on my PSP3000 with v1710012849 pre-release version. I even updated ARK_01234 files and the problem is still there...

krazynez, 2024-03-18:
I tested it on my 3K with the latest pre-release available which includes those patches.

NagaseKouichi, 2024-04-07:
> I tested it on my 3K with the latest pre-release available which includes those patches.

I updated to 4.20.68 R12 and it seems that this issue is fixed.
Thank you.

NagaseKouichi, 2024-04-09:
I find out the problem....It's my bad.

I ran Trick x Logic: Season **1** with XMB but ran Trick x Logic: Season **2** with custom laucher...
They have the same icon and load the same savedata but have different contents.
Sorry for wasting your time.

closed 2024-04-09 by NagaseKouichi