PSP-Archive/ARK-4 #323: pro online with port offset  support?

lightfirst, 2024-03-15:
### New Feature

PPSSPP Support port offset from https://github.com/hrydgard/ppsspp/pull/8496
It there have a pro online patch with port offset  support?
Or someone can move PPSSPP's change to PSP?
pro online for PSP: https://github.com/MrColdbird/aemu

krazynez, 2024-03-15:
port offset is for PC's without admin privleges. PPSSPP has that so you can have port 1025+ avaiable to play online with other PPSSPP users not PSP's.

Double-0-seven7, 2024-03-24:
This is because the default port offset for PPSSPP is now 10000 but this is not really related to the cfw itself.
Need to contact the creators of the plugin or someone that knows it.

krazynez, 2024-03-24:
> This is because the default port offset for PPSSPP is now 10000 but this is not really related to the cfw itself. Need to contact the creators of the plugin or someone that knows it.

Pro online plugin is what it is. Doubt there ever will be an update to it. If you set PPSPPP to 0 offset it matches PSP ports not really sure why you want it to do the same thing the other way around. Maybe ColdBird would be willing to open source pro online that would be the best result.

lightfirst, 2024-03-27:
> Pro online plugin is what it is. Doubt there ever will be an update to it. If you set PPSPPP to 0 offset it matches PSP ports not really sure why you want it to do the same thing the other way [...]

pro online is open source: https://github.com/MrColdbird/aemu

Asteliks, 2025-01-10:
The use case for setting a port offset on PSP arises when you want to play a multiplayer game with one person on a PSP and another on PPSSPP. Currently, you have to set the PPSSPP offset to 0, but this doesn't work for some games, such as Patapon 2. A potential solution could be enabling the option to set a port offset on the PSP itself.

Linked to issue: https://github.com/hrydgard/ppsspp/issues/19842

krazynez, 2025-01-10:
If anyone were to do that it would be a plugin not for ARK-4 to do.

JoseAaronLopezGarcia, 2025-03-16:
Closing this as it's unrelated to ARK. PRO Online is a separate (Open Source) project.

closed 2025-03-16 by JoseAaronLopezGarcia

Asteliks, 2025-03-16:
> Closing this as it's unrelated to ARK. PRO Online is a separate (Open Source) project.

Agree but I think it should be closed as not planned. Otherwise it might be misleading

Kethen, 2025-06-03:
linking this here for those who are still looking for something like this

https://github.com/Kethen/aemu/releases

DoomSlayer9000, 2025-06-04:
> linking this here for those who are still looking for something like this
> 
> https://github.com/Kethen/aemu/releases

While it does work on a psp 3000, it seems to be writing logs all the time which leads to a crash in about 3 minutes when connection is established(can be solved by setting the log file as read only)