PSP-Archive/ARK-4 #324: basic cheat support for PSX games on vita

xor32dotcom, 2024-03-20:
### New Feature

it seems straightforward to at least allow the user to manually add cheats, and it would be preferred if it was added to the menu where you can also remap controllers, enable analog,etc. allowing the user to enter an adress and value they wish to modify

if somebody can reference the source files needed i can try to push it but i doubt i can easily compile this on my own machine as i dont wanna risk bricking any devices :$

xor32dotcom, 2024-03-20:
i found this version of cwcheat apparently form 2008 that worked for me on vita pops
.
https://drive.google.com/file/d/1-PF-CmVGVvW5i-IXFxan4HLXZP8bUrm1/view?pli=1

krazynez, 2024-03-24:
> i found this version of cwcheat apparently from 2008 that worked for me on vita pops . https://drive.google.com/file/d/1-PF-CmVGVvW5i-IXFxan4HLXZP8bUrm1/view?pli=1

Interresting so what happens is you update the cheat.db does it still work or break?

xor32dotcom, 2024-04-26:
> > i found this version of cwcheat apparently from 2008 that worked for me on vita pops . https://drive.google.com/file/d/1-PF-CmVGVvW5i-IXFxan4HLXZP8bUrm1/view?pli=1
> 
> Interresting so what happens is you update the cheat.db does it still work or break?

I accidentally cloned a cheat using it but whatever button combo i pressed updated the cheat.db. I can update the tempAR cheat.db as well and i found that loading bubbles glitches tempAR a little bit but the cheat.db can more or less be updated whenever as long as the plugin interface loads and all