PSP-Archive/ARK-4 #325: Could not Execute Kernel Function

Panzer-II, 2024-03-23:
### PSP/Vita Model

PSP-1000 (Fat)

### ARK Version

ARK-X Version 4.20.68 (2024-02-06)

### Steps to reproduce

Installing Ark X onto PS VIta 1000 following the step in the wiki would produce "Error (FFFFFFFF) : could not execute kernel function"

### USB Charge

Always

### Overclock

Always

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Always

### Memory Stick Speedup

Disabled

### Inferno Cache

LRU

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

On

### Hide DLC

Off

### Turn off LEDs

Disabled

### Verification of form.

- [X] I agree that I properly filled out the items listed above.

krazynez, 2024-03-23:
ARK-X is not for PSP its for PS Vita.

Panzer-II, 2024-03-23:
Yes, I install it on a PS Vita 1000

krazynez, 2024-03-23:
[image]
why did you select the wrong model then?

Panzer-II, 2024-03-23:
just edited

krazynez, 2024-03-23:
Did you reboot once after you installed it? Sometime that helps things.

krazynez, 2024-03-23:
Ohhhhh don't use via Adrenaline, its a standalone applicationl

Panzer-II, 2024-03-25:
To be honest, after I did that the PS Vita crash, and left it to drain before relaunching it, afterwards, it's just work?

closed 2024-03-25 by Panzer-II

JoseAaronLopezGarcia, 2024-03-25:
> To be honest, after I did that the PS Vita crash, and left it to drain before relaunching it, afterwards, it's just work?

You can hold the power button for 10+ seconds to force it to shutdown. Though I'm glad it's working fine now.

Hentilador, 2024-04-12:
I tried installing ark for adrenaline, when launching the ARK Loader i get the "could not execute kernel function" error.

isolatedinsomnia, 2024-09-19:
Sorry to nudge this, but I'm having the same issue.

I'm also on PSV-1000 (Fat), Japan import, and I followed the guide to install Adrenaline here https://vita.hacks.guide/adrenaline.html

I'm on Firmware 3.65 (PSN spoof 3.74) with HENKaku, with PSVSD attached to ux0, tai is read from ur0.

I tried ARK versions:
4.20.68 r14
4.20.69 (DC10) r15
v1726759870

It seems running ARK through FasterARK works, I'm able to boot into the Custom Launcher, but attempting to run ARK_Loader in any way (even through Adrenaline Bubbles) gives me: 

```
LoadingARK -4
Reading kxploit file...
Loading Kxploit at ms0:/PSP/GAME/ARK-Loader/K.BIN
Scanning stubs for kxploit...
Doing kernel exploit...
Escalating privilages...
Scanning kernel functions
Repairing kernel
Autodetecting device
could not detect device, aborting...
ERROR (FFFFFFFF): Could not execute kernel function
```

If there's any more information I can provide, please let me know!