PSP-Archive/ARK-4 #326: Custom Launcher flicker issue while scrolling between tabs

AxolotlAccount, 2024-04-01:
### PSP/Vita Model

PSP-3000 (Slim & Bright)

### ARK Version

Version 4.20.68 Rev 12

### Plugins ( if applicable )

game, ms0:/seplugins/prxshot_png_mod/prxshot_png_mod.prx, on
game, ms0:/seplugins/dayviewer_game.prx, on
game, ms0:/seplugins/npdrm_free_mod.prx, on
game, ms0:/seplugins/nploader_mod.prx, on
vsh, ms0:/seplugins/category_lite.prx, on
vsh, ms0:/seplugins/cxmb/cxmb.prx, off
vsh, ms0:/seplugins/prxshot_png_mod/prxshot_png_mod.prx, on
vsh, ms0:/seplugins/dayviewer.prx, on
vsh, ms0:/seplugins/npdrm_free_mod.prx, on
pops, ms0:/seplugins/cdda_enabler.prx, on
pops, ms0:/seplugins/prxshot_png_mod/prxshot_png_mod.prx, on

### Steps to reproduce

1. Open Custom Launcher from Extras
2. Press Triangle
3. Switch from Games to Network by pressing right from D-pad two times
4. Switch from Network to CFW Settings

When you switch from Network to CFW Settings; Files icon will flicker to Network icon, Network icon will flicker to CFW Settings icon and CFW Settings icon will flicker to Plugins icon temporarily. After that, icons will be back to normal and scroll animation will continue normally.

### USB Charge

Disabled

### Overclock

Always

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Always

### Inferno Cache

LRU

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

On

### Hide DLC

Off

### Turn off LEDs

Disabled

### Verification of form.

- [X] I agree that I properly filled out the items listed above.

AxolotlAccount, 2024-04-15:
This issue still happens with the Version 4.20.68 Rev 14 version of the ARK-4 Firmware.

JoseAaronLopezGarcia, 2024-04-15:
I'm currently busy with a much bigger feature. Stuff like this will have to wait. Maybe @krazynez can take care of this one.

krazynez, 2024-04-21:
This is now in 4.20.69 if you can try it as well so we can close the issue thank you :smile:

AxolotlAccount, 2024-04-21:
It is now fixed, thank you!

closed 2024-04-21 by AxolotlAccount