PSP-Archive/ARK-4 #327: Chinese in the VSH menu does not display properly

li-fstz, 2024-04-04:
### PSP/Vita Model

PSP-3000 (Slim & Bright)

### ARK Version

4.20.68 r12

### Steps to reproduce

1. Choose `简体中文` in `Settings - System Settings - System Language`
[image]
2. Open VSH menu
[image]

### USB Charge

Always

### Overclock

Always

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Always

### Inferno Cache

LRU

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

On

### Hide DLC

Off

### Turn off LEDs

Disabled

### Verification of form.

- [X] I agree that I properly filled out the items listed above.

JoseAaronLopezGarcia, 2024-04-04:
As mentioned here: https://github.com/PSP-Archive/ARK-4/pull/300#issuecomment-2036973787

We need to create font files with chinese characters in the format that the font reader expects.

li-fstz, 2024-04-04:
the code of drawing font is `extras/menus/vshmenu/blit.c`, only work for ascii (or any 8-bit encoding), 
so chinese cannot be read correctly in UTF-8 (japanese and korean also)

krazynez, 2024-04-04:
> the code of drawing font is `extras/menus/vshmenu/blit.c`, only work for ascii (or any 8-bit encoding), so chinese cannot be read correctly in UTF-8 (japanese and korean also)

should be in advancedvshmenu vshmenu is the old menu we do not use. But avshmenu is very much based on that still so probably same issue applies. (More of and FYI)

li-fstz, 2024-04-04:
the `pf` font only support 256 characters, it‘s unlikely to create a standard encoded chinese font

JoseAaronLopezGarcia, 2024-04-04:
> > the code of drawing font is `extras/menus/vshmenu/blit.c`, only work for ascii (or any 8-bit encoding), so chinese cannot be read correctly in UTF-8 (japanese and korean also)
> 
> should be in advancedvsh. vshmenu is the old menu we do not use. But avshmenu is very much based on that still so probably same issue applies. (More of and FYI)

I renamed advancedvsh to vshmenu to keep it standarized with the others.

JoseAaronLopezGarcia, 2024-04-04:
> the code of drawing font is `extras/menus/vshmenu/blit.c`, only work for ascii (or any 8-bit encoding), so chinese cannot be read correctly in UTF-8 (japanese and korean also)

We do have the source code so we might be able to add UTF-8 support to it. Either that or maybe change to the intraFont library so we can load the native PSP fonts as we do in Custom Launcher.

li-fstz, 2024-04-04:
makeing a couple specific encoding satelite_chs.txt and pf font maybe an acceptability idea?
256 may cover all characters in the chinese vsh menu

JoseAaronLopezGarcia, 2024-04-04:
> makeing a couple specific encoding satelite_chs.txt and pf font maybe an acceptability idea? 256 may cover all characters in the chinese vsh menu

So you mean to map ASCII characters to (common/needed) Chinese characters?
We did something similar with the Greek translation in Custom Launcher. I couldn't get intraFont to display Greek characters (even though it should be able to) but I managed to overcome this by creating a special font that mapped Latin characters to Greek.

li-fstz, 2024-04-04:
the idea i'm talking about is more like a temporary solution, the specially satelite_chs.txt and pf-font are not convenient for updating.
the solution you're talking about is more conducive to extend to other unsupported languages.

li-fstz, 2024-04-04:
done
i make a couple pf and txt, it's work

the pf look like:
[image]

[image]
[image]

JoseAaronLopezGarcia, 2024-04-04:
Nice! Open a Pull Request and we will merge it.

closed 2024-04-05 by li-fstz