PSP-Archive/ARK-4 #330: VSH Menu r12

MarcosConci, 2024-04-06:
### PSP/Vita Model

PSP-3000 (Slim & Bright)

### ARK Version

4.20.68r13

### Steps to reproduce

After updating via "ARK-4 Updater" the revision don't change (still r12). Soo there will always be an UPDATE avaliable.

### USB Charge

XMB

### Overclock

Disabled

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

XMB

### Inferno Cache

LRU

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

Off

### Hide DLC

On

### Turn off LEDs

Disabled

### Verification of form.

- [X] I agree that I properly filled out the items listed above.

krazynez, 2024-04-06:
Oh that's my fault forgot to bump version thanks.

krazynez, 2024-04-06:
Fixed. Update again it will show r13. Thanks. Just for verification can you update again and then you or I can close this ticket once verified.

MarcosConci, 2024-04-06:
Awesome work with the CFW. Thanks.

closed 2024-04-06 by MarcosConci