PSP-Archive/ARK-4 #331: Load games from uma0 with standalone mode on vita.

NagaseKouichi, 2024-04-07:
### New Feature

Hello. I'm using sd2vita on my PSV and mount sd2vita to ux0 and memory card to uma0. And I add PSP games to uma0.

In adrenaline mode, I can point the ePSP's ms0 to uma0:/pspemu in settings, so in adrenaline mode ePSP can find psp/ps games in uma0 drive.
But in standalone mode, both ARK and ARK-X only find games from ux0, and I can't find a setting to mount ms0 to uma0:/pspemu.
Is there a way to let it find games from uma0 in standalone mode?

NagaseKouichi, 2024-04-24:
Any chance for this feature?

JoseAaronLopezGarcia, 2024-04-24:
It should be easy to do with a plugin. I'm not sure if it should be part of the ARK repo.

NagaseKouichi, 2024-04-25:
> It should be easy to do with a plugin. I'm not sure if it should be part of the ARK repo.

PSV plugins can have effect with ARK and ARK-X standalone mode? I don't think PSP plugins can do the remount job, because you have to mount ms0: first to have PSP plugins work.
Is there any plugins can already do this?

JoseAaronLopezGarcia, 2024-04-25:
We need a Vita (henkaku) plugin. Similar to how ARK-X needs a plugin to work.

NagaseKouichi, 2024-04-25:
> We need a Vita (henkaku) plugin. Similar to how ARK-X needs a plugin to work.

Adrenaline has this feature build in without need of plugin, maybe that's why there isn't any plugin for this. Hope ARK-4 can add this someday.

JoseAaronLopezGarcia, 2024-04-25:
> > We need a Vita (henkaku) plugin. Similar to how ARK-X needs a plugin to work.
> 
> Adrenaline has this feature build in without need of plugin, maybe that's why there isn't any plugin for this. Hope ARK-4 can add this someday.

Adrenaline does use plugins, it just loads it automatically. Adrenaline patches the native-side of the PS Vita so much that from the (e)PSP side it's almost a completely different system.

Hence why ARK does a technical distinction between Official PSP Emu (Standalone ePSP Installation) and Patched PSP Emu (Adrenaline vPSP Installation).

In any case, the code needed to implement this feature was actually available in ARK-X's plugin (ps1cfw_enabler), I just removed it cause I did't needed it for ARK-X, but it should be very easy to just copy-paste that plugin, and leave only the ms0-redirection code.

NagaseKouichi, 2024-04-27:
> Adrenaline does use plugins, it just loads it automatically. Adrenaline patches the native-side of the PS Vita so much that from the (e)PSP side it's almost a completely different system.
>
> Hence why [...]

Thank you for your great work. :)

JoseAaronLopezGarcia, 2025-01-20:
It's been a while, but just so you know, I decided to implement this feature.
You can now install a native vita plugin called ef2uma.suprx (bundled with ARK) with the latest pre-release.
This will map uma0 as ef0 (as in, PSP Go's internal memory), thus allowing you to use BOTH devices at the same time.

closed 2025-01-20 by JoseAaronLopezGarcia