PSP-Archive/ARK-4 #332: Windows defender alert

MajorTom87, 2024-04-11:
### New Feature

Hi, I downloaded ARK Version 4.20.68 Rev 14 and as soon as it downloaded, it was blocked by my antivirus with the following message: Trojan:Script/Wacatac.H!ml 
I imagine it's a false positive, but it shouldn't give that warning if it's safe...

krazynez, 2024-04-11:
I have no idea what that file is we don't have a Wacatac file at all so something might be trying to inject something when you download or yeah just some weird false positive. If you can get some more details info on that file (safety) would be interesting what is getting flagged, possibly one of our scripts or and exe we have in the contrib/loader folder.

MajorTom87, 2024-04-11:
[image]
Can this help?

krazynez, 2024-04-11:
> ![Cattura](https://private-user-images.githubusercontent.com/78448919/321786823-86495411-8591-452c-8378-383ebd9cfe27.PNG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTI4NzA0NzUsIm5iZiI6MTcxMjg3MDE3NSwicGF0aCI6Ii83ODQ0ODkxOS8zMjE3ODY4MjMtODY0OTU0MTEtODU5MS00NTJjLTgzNzgtMzgzZWJkOWNmZTI3LlBORz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MTElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDExVDIxMTYxNVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTEwZjNmNmNhZDkzYzBlOGViNTllMDI2MTkzMDMwNDY3Njk4OTdkMzc3YmEyZTRhZjgxY2I0YmY3NjUyOTZmNTgmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.lLAKR9wWFoZnpb6LKhWEd0JLQZ2kfR0yrGNxez5OEPI) Can this help?

Yep thanks. Just doesn't like our iso for Vita Standalone. Typical Windows :joy:

closed 2024-04-11 by MajorTom87

JoseAaronLopezGarcia, 2024-04-12:
> Yep thanks. Just doesn't like our iso for Vita Standalone. Typical Windows 😂

We can get rid of it, nobody ever uses it anyways.

krazynez, 2024-04-12:
> We can get rid of it, nobody ever uses it anyways.

True fasterARK took it over.