PSP-Archive/ARK-4 #338: Clicking sound when VSH MENU open

reukiodo, 2024-04-16:
### PSP/Vita Model

PSP-N1000 (PSP Go)

### ARK Version

4.20.68 r14

### Homebrew/GAME (if applicable)

ARMAGEDDON.iso (UMD movie)

### Steps to reproduce

1. place ARMAGEDDON.iso in ms0:/iso/video/
2. use VSH MENU > ADVANCED VSH MENU > ISO VIDEO MOUNT > ARMAGEDDON.iso to select UMD video
3. after it reboots, exit video player
4. on XMB, highlight ARMAGEDDON so that miniclip plays
5. press SELECT to open VSH MENU > ADVANCED VSH MENU
6. listen to clicks playing on speakers

### USB Charge

Always

### Overclock

Disabled

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Disabled

### Inferno Cache

Off

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

Off

### Hide DLC

Off

### Turn off LEDs

Disabled

### Verification of form.

- [X] I agree that I properly filled out the items listed above.

Aneemasta, 2024-12-24:
FW 6.61 ARK 4.20.69 r151 cIPL: Tried to reproduce bug using `Alien (1979).ISO` from https://archive.org/details/UMD-Video-ISO-2002_20240426.
#### Prerequisites:
1. `Alien (1979).ISO` placed into `ms0:/ISO/VIDEO/`
2. `Autoboot Launcher` is set to `Off`
#### Steps:
1. From `XMB` press `Select` to open `VSH Menu`
2. From `VSH Menu` select `Advanced VSH Menu` with `D-Pad` and press `X`
3. From `Advanced VSH Menu` select `ISO Video Mount` with `D-Pad`
4. Switch `ISO Video Mount` from `None` to `Alien (1979).ISO` with `D-Pad` and press `X`
5. Wait for PSP to reboot to `XMB`
6. From `XMB` select `Video/UMD` and wait for preview to begin to play
7. Repeat steps 1 and 2
#### Expected behaviour:
Besides the sound of UMD Video preview, no sound can be heard.
#### Actual behaviour:
✅Same as expected.

JoseAaronLopezGarcia, 2025-03-16:
Closing this.

closed 2025-03-16 by JoseAaronLopezGarcia