PSP-Archive/ARK-4 #339: VSH MENU text corruption from long UMD video names

reukiodo, 2024-04-16:
### PSP/Vita Model

PSP-N1000 (PSP Go)

### ARK Version

4.20.68 r14

### Homebrew/GAME (if applicable)

Curse of the Golden Flower.iso

### Steps to reproduce

1. copy 'Curse of the Golden Flower.iso' to ms0:/iso/video/
2. open VSH MENU > ADVANCED VSH MENU > ISO VIDEO MOUNT > Curse of the Golden Flower.iso
3. notice text corruption after 'Curse of the Golde' outside the VSH MENU block

### USB Charge

Always

### Overclock

Disabled

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Disabled

### Inferno Cache

Off

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

Off

### Hide DLC

Off

### Turn off LEDs

Disabled

### Verification of form.

- [X] I agree that I properly filled out the items listed above.

reukiodo, 2024-04-16:
Also occurs with short filenames with UTF8 (international) characters.

li-fstz, 2024-04-18:
vsh menu doesn't have full font or utf8 encoding support

JoseAaronLopezGarcia, 2024-04-18:
Use the Custom Launcher to mount Video ISO.

reukiodo, 2024-04-18:
Is there a how to use the custom launcher? I don't even know what that is. Is it installed by default or a separate addon?

krazynez, 2024-04-18:
> Is there a how to use the custom launcher? I don't even know what that is. Is it installed by default or a separate addon?

its default under extras category, unless you have a region that does not have extras it will be under game.

reukiodo, 2024-04-18:
Selecting [PREMIUMボｰタブル.iso](http://reukiodo.duckdns.org/software/Games/PSP/ROMs/Video/PREMIUM%e3%83%9c%ef%bd%b0%e3%82%bf%e3%83%96%e3%83%ab.iso) doesn't work from there either...

Aneemasta, 2024-12-24:
#318

JoseAaronLopezGarcia, 2025-03-16:
We have no plans to fix this on VSH Menu, please use Custom Launcher to mount your UMD Video ISOs with special/long names.

closed 2025-03-16 by JoseAaronLopezGarcia