PSP-Archive/ARK-4 #341: Format >128GB cards

reukiodo, 2024-04-17:
### New Feature

It would be nice if Ark4 could natively format >128GB cards to not require using a long workaround guide https://www.reddit.com/r/PSP/comments/13rmlf7/guide_format_a_sd_card_of_128gb_or_more/ even if it has to be formatted from the Recovery menu in order to do so.

krazynez, 2024-04-17:
I know how to do this, however its not good results. The UI will be horribly slow. Even on my 32GB its slllllloooooowwwwwwwww. I had a tester verify it works though.

reukiodo, 2024-04-17:
I think slowness is relative to different users. Some users can put up with the slowness if it means more game choice wherever they go.

krazynez, 2025-03-25:
Closing as not planned ( better adapters are the limitation not ARK-4 )

closed 2025-03-25 by krazynez