PSP-Archive/ARK-4 #342: Remove non-functional XMB menu items

reukiodo, 2024-04-17:
### New Feature

There are several items in the XMB that require internet services that no longer support PSP that could be removed to tidy things up for the future:

[] Extras > T-DMB ? (Korea)
[] Extras > x-Radar Portable ? (Japan)
[] Music > Music Unlimited powered by Qriocity
[] Music > SensMe channels
[] Network > Online Instruction Manuals
[] Network > LocationFree Player ?
[] Network > Skype
[] PlaystationNetwork

And removing their corresponding *.prx files from flash0 can clear up space for some useful features.

krazynez, 2024-04-17:
There is a [plugin](https://www.gamebrew.org/wiki/XMB_Item_Hider_PSP) to do this.

reukiodo, 2024-04-17:
Yes, they can be hidden, though I don't mean to hide unwanted items, I mean to remove non-working items. No one can use these functions anymore as they depend on either nonexistent network services or network services which no longer support the PSP at all (not even with workarounds). Since no one can use them, they are just wasting space both on the XMB and also some *.prx on the firmware flash0 which can be used for other things, like new features.

krazynez, 2024-04-17:
Actually others use those items for other things not just their original functionality. But also the original functionality on those still work though are limited.

reukiodo, 2024-04-17:
For instance, how does the Skype app still work? Are there workarounds to still make it connect and make audio calls?

JoseAaronLopezGarcia, 2024-04-17:
> For instance, how does the Skype app still work? Are there workarounds to still make it connect and make audio calls?

You can replace them with your own apps, with their own icons, that's what people use them for.

reukiodo, 2024-04-17:
So it is difficult to add or remove icons from XMB, but easy to edit what is already present?

li-fstz, 2024-04-18:
vsh menu doesn't have full font support or utf8 encoding

krazynez, 2024-08-27:
> SensMe being on that list is also completely wrong btw, as long as you have the files for the app it still works fine because it never relied on the internet, sp that should also be cut from the list (the only useful thing you'd want to do there is hide the icon as it doesn't rely on any SensMe-specific prx files, given the app itself runs in Game mode).
> 
> Honestly IMO I think the whole list is nonsense and removing stuff from flash0 can only cause trouble if, e.g. booting to OFW using the Home button on CIPL and whatnot, anything on there that is actually unusable nowadays should be hidden/remapped via XMB Icon Manager or equivalent.
> 
> On that note, all I can see on there that actually is unusable is just Music Unlimited (an app, so doesn't waste any flash0 space), Skype (for 2000/3000/N1000 only) and the Online Manuals (just a shortcut to open the browser to the manual URL, although the manuals are still hosted on Sony servers, they are inaccessible on PSP itself due to outdated security standard support in the browser).

So removing the items from flash would most likely brick your PSP still because vshmain.prx is expecting to parse the icons for those files. So that would have to be patched out to skip over.

closed 2024-08-28 by JoseAaronLopezGarcia