PSP-Archive/ARK-4 #343: ZeroVSH Patcher is not labeled as incompatible with 1k

CheezyFriez12, 2024-04-20:
### PSP/Vita Model

PSP-1000 (Fat)

### ARK Version

4.20.69

### Plugins ( if applicable )

ZeroVSH Patcher

### Steps to reproduce

The plugins reccomendations page on the wiki does not mention that ZeroVSH Patcher is incompatible with the 1k models

### USB Charge

Disabled

### Overclock

Disabled

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Disabled

### Inferno Cache

Off

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

Off

### Hide DLC

Off

### Turn off LEDs

Disabled

### Verification of form.

- [X] I agree that I properly filled out the items listed above.

krazynez, 2024-04-20:
Just the clock/calandar is the othet functionality works on 1K.

Aneemasta, 2024-12-24:
> Isn't this something that should be pointed out by the plugin itself?

[It is actually](https://github.com/NightStar3/zerovsh-patcher), and was there by the time this issue was opened:
>From version 0.2, there is support for the psp go's clock and calendar on other psp models (currently psp 1k is not supported)

CheezyFriez12, 2024-12-25:
> [It is actually](https://github.com/NightStar3/zerovsh-patcher), and was there by the time this issue was opened:

The point is to add a disclaimer to the README.md

krazynez, 2024-12-25:
> The point is to add a disclaimer to the README.md

We can add a note, but its really the end users job to read the plugins own readme to know what it is and what its limitations if any are.