PSP-Archive/ARK-4 #347: Possibility of not installing ARK-X and ps1cfw_enabler and arkrightanalog plugins

Dragartosaziados, 2024-04-23:
### New Feature

Is it possible to be offered the chance of not installing ARK-X and ps1cfw_enabler and arkrightanalog plugins? I guess not everyone wants all of them (I don't any).

Dragartosaziados, 2024-04-23:
> Yes this was a planned thing. Just don't have a vita to test with so its on hold for a bit.

Why and how does PS Vita's standalone ARK-4 make a sound when scrolling PSP games even with sound turned off?

What does "Inferno Cache" do and specifically its "LRU" setting? Can it break PSP games?

Clicking "SELECT" does nothing for some seconds when returning to the XMB from Sleep Mode or Custom Launcher, when exiting a game and after restarting the PSP or VSH. The PSP's Memory Stick LED is blinking and only once it finishes I can access the VSH MENU (among other things). I did a full flash install and deleted VSHMENU.PRX from ARK_01234 and it stayed the same.

Also, why can't the Custom Launcher be accessed from the VSH MENU on the PSP?

JoseAaronLopezGarcia, 2024-04-29:
> 
> Why and how does PS Vita's standalone ARK-4 make a sound when scrolling PSP games even with sound turned off?
> 
No idea, I haven't heard of this bug before. You should create a separate thread to report it, don't overwhelm a single thread with multiple different reports, otherwise we will forget about this one and it will get lost among all the other text.

> 
> What does "Inferno Cache" do and specifically its "LRU" setting? Can it break PSP games?
> 
It can help speedup some games that do a lot of arbitrary reads. It can break some games but it's generally stable. The only game I've seen that breaks is marked in the default settings.txt as disabled.

> Clicking "SELECT" does nothing for some seconds when returning to the XMB from Sleep Mode or Custom Launcher, when exiting a game and after restarting the PSP or VSH. The PSP's Memory Stick LED is [...]

This is normal. When returning from sleep mode all I/O operations need to be resumed and it takes a few seconds for the console to return back to normal. It happens worse with UMD games since they are very slow, oftentimes the game simply freezes for a few seconds until the UMD drive is ready again.
There's also other times where VSH Menu refuses to load because there's some other XMB module loaded already (such as the web browser, image viewer, audio/video player, etc), but you can force it to open with L+R+Select (as mentioned in the Wiki).

>
> Why can't the Custom Launcher be accessed from the VSH MENU on the PSP?
> 

Because you can already access it via the Extras section of the XMB or by pressing L+R+Start+Down at any given moment in the XMB or any game. We wanted to reduce redundancy to clean up things and make the CFW simpler. This is the same reason why the original recovery menu was merged into the Custom Launcher and the classic recovery menu was left as the main one.

>
> Can standalone ARK-4 read all PSP DLC without the aid of plugins? And can it read all EBOOT.PBP games?
> 

When using the PS Vita plugin `NoPspEmuDrm`, which is needed if you're using the FasterARK method, then you won't be needing any PSP-side plugin to play DLC and other PSN stuff. If you chose to use ChovySign to make your own bubble then you will have to manually install either `NoPspEmuDrm` on Vita-side, or npdrm_free+nploader on PSP side (either the Vita plugin, or the two PSP plugins, but both at the same time is redundant).

>
> Also, does PSP's ARK-4 need npdrm_free_mod and nploader_mod to read all PSP DLC? And does it need them to read all EBOOT.PBP games as well?
>

Yes, on PSP you need npdrm_free_mod and nploader_mod for compatibility with encrypted PSN content.

Please note that all this (both Vita and PSP) is only for **encrypted** PSN content, if the content is decrypted (which you can do with any PC tool) then you won't need any plugin. Or if you legaly own the thing, then you don't need plugins either.

Dragartosaziados, 2024-04-29:
> No idea, I haven't heard of this bug before. You should create a separate thread to report it, don't overwhelm a single thread with multiple different reports, otherwise we will forget about this [...]

I didn't think it was a bug since I've always had it.

What about the other Inferno Cache setting (not the "LRU" one)?

I was told doing a full flash install and deleting VSHMENU.PRX from ARK_01234 would fix the problem.

What do you mean by encrypted and decrypted PSN content? 

Thanks.

JoseAaronLopezGarcia, 2025-03-16:
There's instructions to (easily) install whatever you want manually.

closed 2025-03-16 by JoseAaronLopezGarcia