PSP-Archive/ARK-4 #348: ARK_cIPL Not Installing On PSP Street  E1003 Hence Cannot Install Permanent CFW

everyonesweird, 2024-04-23:
After running live Ark, I tried to install cIPL using the flasher. The flasher installs and then reboots, but I do not see cIPL when I go check it in the system information. Hence I cannot install Ark permanently.
I tried using infinity but the psp shuts down when I load the infinity instead of showing the infinity menu.
I'm using PSP Street E1003 model on 6.61 OFW and latest Ark Version 4.20.69 (DC10).

JoseAaronLopezGarcia, 2024-04-29:
You need to do a full reboot for cIPL to take effect. We should change the installer to do that instead of doing a soft reboot.

krazynez, 2024-04-29:
fixed in https://github.com/PSP-Archive/ARK-4/releases/download/v1714397698/ARK-4-v1714397698.zip

This will now fully reboot the PSP after flashing cIPL or restoring IPL.

everyonesweird, 2024-04-30:
> fixed in https://github.com/PSP-Archive/ARK-4/releases/download/v1714397698/ARK-4-v1714397698.zip
> 
> This will now fully reboot the PSP after flashing cIPL or restoring IPL.

Thank you all. Its now Installed properly and permanently.

closed 2024-04-30 by krazynez