PSP-Archive/ARK-4 #350: ARK-4 Plugin Manager showing duplicated entries

pebeto, 2024-04-23:
### PSP/Vita Model

PSP-1000 (Fat)

### ARK Version

ARK-4.20.69

### Plugins ( if applicable )

```
vsh, ms0:/seplugins/RemoteJoyLite.prx, 1
game, ms0:/seplugins/RemoteJoyLite.prx, 1
```

### Steps to reproduce

1. Manually add any plugin into your `PLUGINS.TXT` file under `SEPLUGINS`
2. Ensure you have an entry for `VSH` and `GAME`
3. Reset your device, and check `Plugins Manager` (it can be seen without resetting)

### USB Charge

Disabled

### Overclock

Disabled

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Always

### Inferno Cache

LRU

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

On

### Hide DLC

Off

### Turn off LEDs

Disabled

### Verification of form.

- [X] I agree that I properly filled out the items listed above.

krazynez, 2024-04-23:
Its not duplicated one is for VSH and the other is for Game?

pebeto, 2024-04-23:
https://github.com/PSP-Archive/ARK-4/assets/28990958/4ece89bd-f8c0-43fb-9ed8-24e88cd55260

krazynez, 2024-04-23:
Its not duplicated one is for VSH and the other is for Game? 

> @krazynez
> ARK_Plugin_Duplication.mp4

Ahhh I see interesting.

pebeto, 2024-04-23:
The issue could come from here

https://github.com/PSP-Archive/ARK-4/blob/2963601d2ea0fd5e2a6e5bc3a130297340ed837b/extras/menus/arkMenu/include/ark_plugins.h#L93

JoseAaronLopezGarcia, 2024-04-24:
Makes sense. Confirm it's fixed and close issue.

JoseAaronLopezGarcia, 2024-04-29:
Closing as it should be fixed.

closed 2024-04-29 by JoseAaronLopezGarcia