PSP-Archive/ARK-4 #351: DC10 - Physical formatting of PSP Go mass storage

MRiCEQB, 2024-04-24:
### New Feature

I have a PSP Go, where the NAND has suffered severe deterioration over time, in form of bad blocks.
It started with corrupted save games and got to a point, where the PSP was requesting me to format the system storage upon boot.

After some time and multiple reformats later, it wasn't able to recover the system storage again and just gave out an error while trying to format it again (Format failed. (802E110A)).

It got worse and worse over the span of a few months, until the PSP didn't boot anymore.
I read online that people fixed their regular PSPs bad blocks with DC in the past, thanks to the "Format LFlash" feature, which probably causes the PSP (or the NAND controller) to skip bad blocks after formatting.

Thanks to your work on DC10 and Baryon Sweeper, I was able to ressurrect this PSP Go from the dead by formatting the LFLash and doing a clean install of the FW.
Unfortunately the system storage formatting error remained and there is nothing I can do as a user to fix it with the available software.

My first idea was to replace the NAND, but the part seems to be custom made for the PSP Go and prices for spare parts (in form of defective boards/consoles) have risen over time, since the PSP Go became a rare system, due to its short life span and low sales.

My last hope is this feature request here, with which I would like to suggest an additional option in the "NAND Operations" menu, which enables the user to do a physical format of the system storage (ef0:), in the same way that the "FLash Format" feature formats the LFlash.
My hope is that such a physical format will signal the NAND (or its Controller), to skip the bad blocks in the system storage and allow the PSP to format the space again.
If such a feature is feasible to implement into DC10, I would be forever grateful.

Please let me know what you think about it, or if you need further information or if you have anything to test.

Thank you and please keep up the great work!

krazynez, 2024-04-24:
NAND and ef0 are not on the same chipset formatting ef0 would not do anything. You can (back up your nand first) adjust the lflash values and try to force a format of the bad blocks with adjusted values. Then resize them to defaults and format again. Bad blocks are expected though so some bad blocks are just from manufacturing defects.

MRiCEQB, 2024-04-24:
I just took a look at it, but I can't go higher than the default values in the LFlash formatting settings.
I can only manipulate the sizes within that 64MB window :(

krazynez, 2024-04-24:
> I just took a look at it, but I can't go higher than the default values in the LFlash formatting settings. I can only manipulate the sizes within that 64MB window :(

yeah..... Go only has 64MB I meant make the values smaller to non standard 512 bs to try to overwrite the bad sectors.

MRiCEQB, 2024-04-24:
But that wouldn't help me if I'm not mistaken.
Because the OS side of things works now.
What I want to do is do a physical format of the system storage area of the flash chip.

My hope is that skips the bad blocks I have in that area which (probably) prevent me from formatting it within the VSH.

My understanding of the NAND layout is like this:

1) IPL
2) IDStorage
3) flash0
4) flash1
5) flash2
6) flash3
7) System Storage (PSP Go only).

It would be awesome if the LFlash formatting could  be expanded to go past the flash3 barrier.

krazynez, 2024-04-24:
> But that wouldn't help me if I'm not mistaken. Because the OS side of things works now. What I want to do is do a physical format of the system storage area of the flash chip.
>
> My hope is that skips [...]

Okay your not really getting what nand is it seem, however adding formatting ef0 would produce the same results that xmb does as we use Sony syscalls for all functionality.

JoseAaronLopezGarcia, 2024-04-24:
> Okay your not really getting what nand is it seem, however adding formatting ef0 would produce the same results that xmb does as we use Sony syscalls for all functionality.

He has a point though, I always thought ef0 was part of nand in PSP Go. It makes sense, why have two separate chips.

MRiCEQB, 2024-04-24:
> Okay your not really getting what nand is it seem, however adding formatting ef0 would produce the same results that xmb does as we use Sony syscalls for all functionality.

I'm no expert you are correct :(
But I believe what you are saying.
So there are no syscall(s) available to resize/repartition ef0?

MRiCEQB, 2024-04-24:
> He has a point though, I always thought ef0 was part of nand in PSP Go. It makes sense, why have two separate chips.

Is it not tho?
Looking at the PCB, I only see one flash chip.

krazynez, 2024-04-24:
> I'm no expert you are correct :( But I believe what you are saying. So there are no syscall(s) available to resize/repartition ef0?

there is to format but not resize, though I wonder what would happen if it was formatted as fat16 instead of 32 would be interesting.

MRiCEQB, 2024-04-24:
> there is to format but not resize, though I wonder what would happen if it was formatted as fat16 instead of 32 would be interesting.

If you can implement that, I'm happy to try it :D
Anything that touches/formats the system strage I'm willing to try.

krazynez, 2024-05-02:
> If you can implement that, I'm happy to try it :D Anything that touches/formats the system strage I'm willing to try.

Are you in the discord?

krazynez, 2024-05-04:
Implemented via PSP Tool, however due to eMMC being fully corrupted will no longer format successfully.

Aldighieri22, 2025-02-25:
Hi, i have a PSPgo and saved it from a hard brick using the pysweeper software, but i have the same problem as @MRiCEQB, PSPgo cant format internal memory, and its very slow too boot. Any solution? 
@krazynez do you know the IC that have the psp firmware? 

Sry for my english.

krazynez, 2025-02-25:
> Hi, i have a PSPgo and saved it from a hard brick using the pysweeper software, but i have the same problem as @MRiCEQB, PSPgo cant format internal memory, and its very slow too boot. Any solution? 
> @krazynez do you know the IC that have the psp firmware? 
> 
> Sry for my english.

You would need to either remove and readd a new eMMC or just swap motherboards. Its a hardware failure no software will fix it.

Aldighieri22, 2025-02-25:
> You would need to either remove and readd a new eMMC or just swap motherboards. Its a hardware failure no software will fix it.

You know if i can remove eMMC? PSP works without the emmc ci?

krazynez, 2025-02-25:
> You know if i can remove eMMC? PSP works without the emmc ci?

No idea never tried.

Aldighieri22, 2025-02-25:
> No idea never tried.

One last question, you know the datasheet for this emmc? I tried to find it but no luck with the part number.

Aldighieri22, 2025-02-25:
Good news!! I just downgrade my PSPgo from 6.61 to 6.60 and internal memory is fine again!! Idk if the unbrick metode corrupt something, but now its working perfect.

MRiCEQB, 2025-02-25:
> I just downgrade my PSPgo from 6.61 to 6.60 and internal memory shows again, but when i try to install cfw its corrupt again.
> 
> I think the best alternative is a way to disable the internal memory, so pspgo will stop to try to acess this memory.

Did you use Chronoswitch to downgrade?

Aldighieri22, 2025-02-25:
> > I just downgrade my PSPgo from 6.61 to 6.60 and internal memory shows again, but when i try to install cfw its corrupt again.
> > I think the best alternative is a way to disable the internal memory, so pspgo will stop to try to acess this memory.
> 
> Did you use Chronoswitch to downgrade?

Yes, very easy to use, but its not the problem, the best solution would be to disable the internal memory, so PSP could stop to try to acess this memory. My PSP took 1 minute more to turn on, turn off, open or close games, just because he try everytime to acess the internal memory. :(

Aldighieri22, 2025-02-26:
> > I just downgrade my PSPgo from 6.61 to 6.60 and internal memory shows again, but when i try to install cfw its corrupt again.
> > I think the best alternative is a way to disable the internal memory, so pspgo will stop to try to acess this memory.
> 
> Did you use Chronoswitch to downgrade?

Do you tried to downgrade? Its worked?

MRiCEQB, 2025-03-01:
> Do you tried to downgrade? Its worked?

Downgrading doesn't work, it always tells me that it couldn't delete the Resume Game data and quits the app for safety reasons.
This happens on every version of Chronoswitch, even on 7.0 by Davee.
Upgrading on the other hand works but the error still remains after a reboot, where the internal memory cannot be formatted.

krazynez, 2025-03-01:
Yes because its trying to read eflash0 and can't so it quits

Aldighieri22, 2025-03-02:
> Yes because its trying to read eflash0 and can't so it quits

Krazynez, give us a option to skip ef0 acess on pspgo when ef0 are broken. 😁

JoseAaronLopezGarcia, 2025-03-18:
> > Yes because its trying to read eflash0 and can't so it quits
> 
> Krazynez, give us a option to skip ef0 acess on pspgo when ef0 are broken. 😁

Latest release provides just that. Closing this thread.

closed 2025-03-18 by JoseAaronLopezGarcia