PSP-Archive/ARK-4 #354: Turn off LEDs Glitch

Toddimyre, 2024-04-27:
### PSP/Vita Model

PSP-2000 (Slim)

### ARK Version

4.20.69 r2

### Plugins ( if applicable )

game, ms0:/SEPLUGINS/npdrm_free_mod.prx, on
game, ms0:/SEPLUGINS/nploader_mod.prx, on
vsh, ms0:/SEPLUGINS/dlmgr.prx, on
vsh, ms0:/SEPLUGINS/hold.prx, on

### Steps to reproduce

1. Power on PSP.
2. Scroll left to Extras.
3. Select Custom Firmware Settings.
4. Scroll down to Turn off LEDs.
5. Select an option other than Disabled.
6. Press Select and Reset VSH. -Optional
7. Load whatever option selected above to turn off the LED as intended.
8. Put the device into sleep mode.
9. Wake the device.

### USB Charge

Disabled

### Overclock

Disabled

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Disabled

### Inferno Cache

LRU

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

On

### Hide DLC

On

### Turn off LEDs

Always

### Verification of form.

- [X] I agree that I properly filled out the items listed above.

hb-webdev, 2025-02-04:
> We are looking into ways around that with power handlers and such to see if we can indeed leave them disabled upon resuming from sleep.

Does this part of the code help from [a plugin I updated](https://archive.org/details/power-led-disabler) for PSP Go a few years back?

https://ia800702.us.archive.org/12/items/power-led-disabler/main.c

```
while ( 1 )  // We do this because if we didn't put it, suspending the PSP and turning it back on would turn on the power LED 
{ 
sceSysconCtrlLED ( MS_LED , STATE_OFF ) ;  // Turn off the Memory Stick access LED 
sceSysconCtrlLED ( WLAN_LED , STATE_OFF ) ;  // We turn off the WLAN LED 
sceSysconCtrlLED ( POWER_LED , STATE_OFF ) ;  // We turn off the POWER LED (this does not affect if the power cable is connected) 
sceSysconCtrlLED ( BT_LED , STATE_OFF) ; // Turn off the BlueTooth LED
}
```

wad11656, 2025-02-04:
> We are looking into ways around that with power handlers and such to see if we can indeed leave them disabled upon resuming from sleep.

@krazynez Does this part of the code help from [a plugin I updated](https://archive.org/details/power-led-disabler) for PSP Go a few years back?

https://ia800702.us.archive.org/12/items/power-led-disabler/main.c

```
while ( 1 )  // We do this because if we didn't put it, suspending the PSP and turning it back on would turn on the power LED 
{ 
sceSysconCtrlLED ( MS_LED , STATE_OFF ) ;  // Turn off the Memory Stick access LED 
sceSysconCtrlLED ( WLAN_LED , STATE_OFF ) ;  // We turn off the WLAN LED 
sceSysconCtrlLED ( POWER_LED , STATE_OFF ) ;  // We turn off the POWER LED (this does not affect if the power cable is connected) 
sceSysconCtrlLED ( BT_LED , STATE_OFF) ; // Turn off the BlueTooth LED
}
```

JoseAaronLopezGarcia, 2025-02-04:
That code is way too unoptimized for us to include it in ARK. It's far better to just register a power callback and detect resume events to turn LEDs off again once resume is complete.

JoseAaronLopezGarcia, 2025-03-16:
Fixed on latest (pre)release.

closed 2025-03-16 by JoseAaronLopezGarcia