PSP-Archive/ARK-4 #356: DC 10 INSTALLER

samfadel, 2024-04-29:
### PSP/Vita Model

PSP-2000 (Slim)

### ARK Version

ARK 4.20.69 r2

### Steps to reproduce

  'Use Team C+D mspformat pc tool and run the installer again' appear when I try to use DC10 installer

### USB Charge

Always

### Overclock

Disabled

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Disabled

### Inferno Cache

Off

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

Off

### Hide DLC

Off

### Turn off LEDs

Disabled

### Verification of form.

- [X] I agree that I properly filled out the items listed above.

krazynez, 2024-04-29:
You need to format your memory stick to have enough space for the IPL, you can use PSP Tool to format it if need be.

samfadel, 2024-04-29:
My psp memory card (offical) has 2 gb free space

krazynez, 2024-04-29:
Free space has nothing to do with the ipl sector that goes after the MBR not the partition table.

closed 2024-04-29 by samfadel

JoseAaronLopezGarcia, 2024-04-29:
We really should work on that formatting part of the PC installer.

samfadel, 2024-05-01:
I fix it
1- backup my memory stick files 
2- format memory stick using psp tool
3- use DC installer ( put 661.PBP on root memory stick
4- after installing copy backup files

krazynez, 2024-05-01:
> I fix it 1- backup my memory stick files 2- format memory stick using psp tool 3- use DC installer ( put 661.PBP on root memory stick 4- after installing copy backup files

Newest ipl should be small enough to not have to ever format again. But glad you resolved it.

JoseAaronLopezGarcia, 2024-05-01:
> > I fix it 1- backup my memory stick files 2- format memory stick using psp tool 3- use DC installer ( put 661.PBP on root memory stick 4- after installing copy backup files
> 
> Newest ipl should be small enough to not have to ever format again. But glad you resolved it.

Some cards come preformated without an MBR at all (FAT takes up all of the device, no partitions) so you can't inject IPL since they have 0 space for them.
A simple formatting using the PSP's own formatter would have fixed it.