PSP-Archive/ARK-4 #358: Boot 6.61 ARK from  memory stick

samfadel, 2024-04-30:
### PSP/Vita Model

PSP-2000 (Slim)

### ARK Version

ARK 4.20.69 r3

### Steps to reproduce

when try to Boot 6.61 ARK from  memory stick go into black screen and turn off

### USB Charge

Always

### Overclock

Disabled

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Disabled

### Inferno Cache

Off

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

Off

### Hide DLC

On

### Turn off LEDs

Disabled

### Verification of form.

- [X] I agree that I properly filled out the items listed above.

krazynez, 2024-04-30:
Thank you we are looking into it.

JoseAaronLopezGarcia, 2024-05-01:
Fixed in Rev 4

closed 2024-05-01 by JoseAaronLopezGarcia