PSP-Archive/ARK-4 #359: PSP's/PS Vita's AR-4 updater and XMB often don't detect new ARK-4 versions

Dragartosaziados, 2024-05-01:
### PSP/Vita Model

PSP-3000 (Slim & Bright), PCH-2000 (Vita Slim), VTE-1000 (Vita TV)

### ARK Version

4.20.69 Rev 4

### Steps to reproduce

PSP's/PS Vita's ARK-4 updater and the XMB often still don't detect a new ARK-4 version or install the same version even when a new version is out. When that happens, I have to manually update it.

### USB Charge

Disabled

### Overclock

Disabled

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Disabled

### Inferno Cache

Off

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

Off

### Hide DLC

Off

### Turn off LEDs

Disabled

### Verification of form.

- [X] I agree that I properly filled out the items listed above.

JoseAaronLopezGarcia, 2024-05-01:
That's because we have to manually update the server. Should be fixed now.

Dragartosaziados, 2024-05-01:
> That's because we have to manually update the server. Should be fixed now.

Does it mean it will keep happening? Also, I don't understand why it sometimes detects the new version if closing and opening the Custom Launcher/ARK-4 and/or enabling the "Force update" setting.

krazynez, 2024-05-01:
> > That's because we have to manually update the server. Should be fixed now.
> 
> Does it mean it will keep happening? Also, I don't understand why it sometimes detects the new version if closing and opening the Custom Launcher/ARK-4 and/or enabling the "Force update" setting.

Well if you "Force Update" it will always think there is a new version. Yes it can happen as @JoseAaronLopezGarcia stated we have to manually update the server to do stable updates its not automated like testing is.

Dragartosaziados, 2024-05-01:
> Well if you "Force Update" it will always think there is a new version. Yes it can happen as @JoseAaronLopezGarcia stated we have to manually update the server to do stable updates its not automated [...]

Even if enabling the "Force update" setting, it will still often not detect updates when they're available but there are times it detects new ones and install them, unlike if the setting is disabled.

krazynez, 2024-05-01:
> > > > That's because we have to manually update the server. Should be fixed now.
> > > 
> > > 
> > > Does it mean it will keep happening? Also, I don't understand why it sometimes detects the new version if closing and opening the Custom Launcher/ARK-4 and/or enabling the "Force update" setting.
> > 
> > 
> > Well if you "Force Update" it will always think there is a new version. Yes it can happen as @JoseAaronLopezGarcia stated we have to manually update the server to do stable updates its not automated like testing is.
> 
> Even if enabling the "Force update" setting, it will still often not detect updates when they're available but there are times it detects new ones and installs them, unlike if the setting is disabled.

Well its best to leave that disabled anyway.

Dragartosaziados, 2024-05-01:
> Well its best to leave that disabled anyway.

Why, since even though it installs the same version (and not the new one, if out), it sometimes installs the new one (when if disabled it sometimes doesn't)?

krazynez, 2024-05-01:
> > > > > > That's because we have to manually update the server. Should be fixed now.
> > > > > 
> > > > > 
> > > > > Does it mean it will keep happening? Also, I don't understand why it sometimes detects the new version if closing and opening the Custom Launcher/ARK-4 and/or enabling the "Force update" setting.
> > > > 
> > > > 
> > > > Well if you "Force Update" it will always think there is a new version. Yes it can happen as @JoseAaronLopezGarcia stated we have to manually update the server to do stable updates its not automated like testing is.
> > > 
> > > 
> > > Even if enabling the "Force update" setting, it will still often not detect updates when they're available but there are times it detects new ones and installs them, unlike if the setting is disabled.
> > 
> > 
> > Well its best to leave that disabled anyway.
> 
> Why, since even though it installs the same version (and not the new one, if out), it sometimes installs the new one (when if disabled it sometimes doesn't), if out?

Only time its really required is if we update the zip and not the revision number. The PSP will check the version string and see its the same number and say there is no update. Even though techinally there is one, just we didn't do a rev push.

Dragartosaziados, 2024-05-01:
> Only time its really required is if we update the zip and not the revision number. The PSP will check the version string and see its the same number and say there is no update. Even though [...]

What do you mean by the first sentence?

Dragartosaziados, 2024-05-01:
> > > > > > > > > That's because we have to manually update the server. Should be fixed now.
> > > > > > > > 
> > > > > > > > 
> > > > > > > > Does it mean it will keep happening? Also, I don't understand why it sometimes detects the new version if closing and opening the Custom Launcher/ARK-4 and/or enabling the "Force update" setting.
> > > > > > > 
> > > > > > > 
> > > > > > > Well if you "Force Update" it will always think there is a new version. Yes it can happen as @JoseAaronLopezGarcia stated we have to manually update the server to do stable updates its not automated like testing is.
> > > > > > 
> > > > > > 
> > > > > > Even if enabling the "Force update" setting, it will still often not detect updates when they're available but there are times it detects new ones and installs them, unlike if the setting is disabled.
> > > > > 
> > > > > 
> > > > > Well its best to leave that disabled anyway.
> > > > 
> > > > 
> > > > Why, since even though it installs the same version (and not the new one, if out), it sometimes installs the new one (when if disabled it sometimes doesn't), if out?
> > > 
> > > 
> > > Only time its really required is if we update the zip and not the revision number. The PSP will check the version string and see its the same number and say there is no update. Even though techinally there is one, just we didn't do a rev push.
> > 
> > 
> > What do you mean by the first sentence?
> 
> ![image](https://private-user-images.githubusercontent.com/39999447/327137798-bef6aec1-ca13-46ec-8a94-13dcdf7ab745.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTQ1NzI5MzIsIm5iZiI6MTcxNDU3MjYzMiwicGF0aCI6Ii8zOTk5OTQ0Ny8zMjcxMzc3OTgtYmVmNmFlYzEtY2ExMy00NmVjLThhOTQtMTNkY2RmN2FiNzQ1LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MDElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTAxVDE0MTAzMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTM4MmI0NGM0OGRjMzkzMWJhMGFhNTQxNmY5YmU5MjA3ZDIzNTg5ZmMyNDhmZjljZjg0YzU3MmJiZTdjNDIzOGUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.ZWGSwyA3_fk1oNTm8MnqQ6rdZAeSMl36mNY5Wr-iZ9Q) That is the zip that the server grabs to extract the update file from. But we use Sony's way to parse if there are new updates from ark-4.ddns.net i.e: http://ark-4.ddns.net/psp-updatelist.txt if the ImageVersion doesn't change (make it highter) the PSP/Vita will think its still the latest version. We have revistions in a global file that update when we update it. i.e:
> 
> https://github.com/PSP-Archive/ARK-4/blob/5dd06e6a788631e713b040bc42df80fc385310c2/common/include/globals.h#L74-L77
> 
> if we do not update the revision or version number but we update the zip file (first image) then the update server will always think its still the latest version available. So having "force update" is good to have when that rarity case happens. It is usually a quick fix before it affects the masses and not worth another revision for a simple fix.

I guess I'll disable it, then (even if sometimes ot can update ARK-4 unlike if disabled).

Where is the settings.txt file located to the "Inferno Cache" setting and what'a the difference between Infermo Cache's "LRU" and "RR" settings?

Also, can the "Memory Stick speedup" setting break PSP games (I'm excluding [this](https://github.com/PSP-Archive/ARK-4/issues/346) issue which affects ARK-4 itself and may affect all PSP games).

krazynez, 2024-05-01:
> > > > > > > > > > That's because we have to manually update the server. Should be fixed now.
> > > > > > > > > 
> > > > > > > > > 
> > > > > > > > > Does it mean it will keep happening? Also, I don't understand why it sometimes detects the new version if closing and opening the Custom Launcher/ARK-4 and/or enabling the "Force update" setting.
> > > > > > > > 
> > > > > > > > 
> > > > > > > > Well if you "Force Update" it will always think there is a new version. Yes it can happen as @JoseAaronLopezGarcia stated we have to manually update the server to do stable updates its not automated like testing is.
> > > > > > > 
> > > > > > > 
> > > > > > > Even if enabling the "Force update" setting, it will still often not detect updates when they're available but there are times it detects new ones and installs them, unlike if the setting is disabled.
> > > > > > 
> > > > > > 
> > > > > > Well its best to leave that disabled anyway.
> > > > > 
> > > > > 
> > > > > Why, since even though it installs the same version (and not the new one, if out), it sometimes installs the new one (when if disabled it sometimes doesn't), if out?
> > > > 
> > > > 
> > > > Only time its really required is if we update the zip and not the revision number. The PSP will check the version string and see its the same number and say there is no update. Even though techinally there is one, just we didn't do a rev push.
> > > 
> > > 
> > > What do you mean by the first sentence?
> > 
> > 
> > ![image](https://private-user-images.githubusercontent.com/39999447/327137798-bef6aec1-ca13-46ec-8a94-13dcdf7ab745.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTQ1NzI5MzIsIm5iZiI6MTcxNDU3MjYzMiwicGF0aCI6Ii8zOTk5OTQ0Ny8zMjcxMzc3OTgtYmVmNmFlYzEtY2ExMy00NmVjLThhOTQtMTNkY2RmN2FiNzQ1LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MDElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTAxVDE0MTAzMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTM4MmI0NGM0OGRjMzkzMWJhMGFhNTQxNmY5YmU5MjA3ZDIzNTg5ZmMyNDhmZjljZjg0YzU3MmJiZTdjNDIzOGUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.ZWGSwyA3_fk1oNTm8MnqQ6rdZAeSMl36mNY5Wr-iZ9Q) That is the zip that the server grabs to extract the update file from. But we use Sony's way to parse if there are new updates from ark-4.ddns.net i.e: http://ark-4.ddns.net/psp-updatelist.txt if the ImageVersion doesn't change (make it highter) the PSP/Vita will think its still the latest version. We have revistions in a global file that update when we update it. i.e:
> > https://github.com/PSP-Archive/ARK-4/blob/5dd06e6a788631e713b040bc42df80fc385310c2/common/include/globals.h#L74-L77
> > 
> > if we do not update the revision or version number but we update the zip file (first image) then the update server will always think its still the latest version available. So having "force update" is good to have when that rarity case happens. It is usually a quick fix before it affects the masses and not worth another revision for a simple fix.
> 
> I guess I'll disable it, then (even if sometimes it can update ARK-4 unlike if disabled).
> 
> Where is the settings.txt file located to the "Inferno Cache" setting and what's the difference between Inferno Cache's "LRU" and "RR" settings?
> 
> Can the "Memory Stick speedup" setting break PSP games (I'm excluding [this](https://github.com/PSP-Archive/ARK-4/issues/346) issue which affects ARK-4 and may affect all PSP games run from it).
> 
> I enabled the "Scan SAVEDATA" setting and savedata still doesn't appear in ARK-4.
> 
> Also, can the "Force Extra Memory" setting break PSP games?
> 
> EDIT: I have issues with the european Spider-Man: Web of Shadows – Amazing Allies PSP ISO on PSP's ARK-4 if the "Force Extra Memory" setting is set to "Always" because the PSP freezes when exiting it using the PS button and it restarts some seconds after.

Please refrain from adding questions not related to the issue this is about, if you wish to ask general questions we have a discord server that would be best to ask there.

Dragartosaziados, 2024-05-01:
> Please refrain from adding questions not related to the issue this is about, if you wish to ask general questions we have a discord server that would be best to ask there.

I don't have a discord account and should I create issues for the problem, then?

JoseAaronLopezGarcia, 2025-03-16:
Closing this.

closed 2025-03-16 by JoseAaronLopezGarcia

MasterTJ123, 2025-10-26:
I've also been having this problem for quite a while. I always have to update manually since the updater doesn’t detect the new releases. Is this still because of the time it takes you to update the server or is something else going on?