PSP-Archive/ARK-4 #360: "Scan SAVEDATA" setting not showing savedata

Dragartosaziados, 2024-05-01:
### PSP/Vita Model

PSP-3000 (Slim & Bright), PCH-2000 (Vita Slim), VTE-1000 (Vita TV)

### ARK Version

4.20.69 Rev 4

### Steps to reproduce

I enabled the "Scan SAVEDATA" setting and savedata still doesn't appear in ARK-4.

### USB Charge

Disabled

### Overclock

Disabled

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Disabled

### Inferno Cache

Off

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

Off

### Hide DLC

Off

### Turn off LEDs

Disabled

### Verification of form.

- [X] I agree that I properly filled out the items listed above.

krazynez, 2024-05-01:
```c
            while ((savedit = readdir(savedir))){
  0                 if (strcmp(savedit->d_name, ".") == 0) continue;
  1                 if (strcmp(savedit->d_name, "..") == 0) continue;
  2                 string fullentrypath = fullpath + "/" + string(savedit->d_name);
  3                 if (Iso::isISO(fullentrypath.c_str())) this->categories[GAME]->addEntry(new Iso(fullentrypath));
  4                 else if (Eboot::isEboot(fullentrypath.c_str())){
  5                     Eboot* e = new Eboot(fullentrypath);
  6                     switch (Eboot::getEbootType(fullentrypath.c_str())){
  7                     case TYPE_PSN:         this->categories[GAME]->addEntry(e);        break;
  8                     case TYPE_POPS:        this->categories[POPS]->addEntry(e);        break;
  9                     default:               this->categories[HOMEBREW]->addEntry(e);    break;
 10                     }
 11                 }
 12             }
 13             closedir(savedir);
```
This checks for ISO/EBOOT/POPS's in SAVEDATA not displaying actual savedata

Dragartosaziados, 2024-05-01:
> ```c
>             while ((savedit = readdir(savedir))){
>   0                 if (strcmp(savedit->d_name, ".") == 0) continue;
>   1                 if (strcmp(savedit->d_name, "..") == 0) continue; [...]

I thought they always needed to be in their respective folders, not in the SAVEDATA one.

krazynez, 2024-05-01:
> I thought they always needed to be in their respective folders, not in the SAVEDATA one.

Believe Acid added that for older style of exploits and such initially similar to ARK where it has ARK_01234 with 99% of our files.

JoseAaronLopezGarcia, 2024-05-01:
This feature scans for ISO and PBP files inside savedata folders, which is an obsolete method of injecting games and homebrew via (Q)CMA on PS Vita.

Dragartosaziados, 2024-05-01:
> This feature scans for ISO and PBP files inside savedata folders, which is an obsolete method of injecting games and homebrew via (Q)CMA on PS Vita.

If it's obsolete, then why wasn't it removed?

JoseAaronLopezGarcia, 2024-05-02:
> > This feature scans for ISO and PBP files inside savedata folders, which is an obsolete method of injecting games and homebrew via (Q)CMA on PS Vita.
> 
> If it's obsolete, then why wasn't it removed? Anyway, the repo should tell what does this feature do.

Because it's easier to not do anything than to remove the feature and test that nothing has broken.

tldr: I'm too lazy to remove it.

Dragartosaziados, 2024-05-03:
> > > This feature scans for ISO and PBP files inside savedata folders, which is an obsolete method of injecting games and homebrew via (Q)CMA on PS Vita.
> > 
> > 
> > If it's obsolete, then why wasn't it removed? Anyway, the repo should tell what does this feature do.
> 
> Because it's easier to not do anything than to remove the feature and test that nothing has broken.
> 
> tldr: I'm too lazy to remove it.
> 
> Also technically speaking, you can and should be able to use ARK on OFW PS Vita with no native hacks, where this feature does become needed.

I want to keep my PS Vita modded and I didn't even knew this setting was for unmodded ones, even though it also works in modded ones.