PSP-Archive/ARK-4 #361: "Force Extra Memory" freezes a PSP game and makes the PSP restart some seconds after

Dragartosaziados, 2024-05-01:
### PSP/Vita Model

PSP-3000 (Slim & Bright)

### ARK Version

4.20.69 Rev 4

### Steps to reproduce

If "Force Extra Memory" is set to "Always", it makes the european Spider-Man: Web of Shadows – Amazing Allies PSP ISO on PSP's ARK-4 freeze when exiting it using the PS button and the PSP restarts some seconds after.

### USB Charge

Disabled

### Overclock

Disabled

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Disabled

### Inferno Cache

Off

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

Off

### Hide DLC

Off

### Turn off LEDs

Disabled

### Verification of form.

- [X] I agree that I properly filled out the items listed above.

krazynez, 2024-05-01:
Well this one is a simple fix, 

Force Extra Memory is only for PRO Online and some Homebrew that require it. Games that utilize it will always have access to it. Its best to only enable this for homebrew/plugins that need it not to always have it enabled.

Dragartosaziados, 2024-05-01:
> Well this one is a simple fix,
> 
> Force Extra Memory is only for PRO Online and some Homebrew that require it. Games that utilize it will always have access to it. Its best to only enable this for homebrew/plugins that need it not to always have it enabled.

If games that use PRO Online will always have access to it, then it doesn't make sense if the setting is disabled and it still works.

krazynez, 2024-05-01:
> If games that use PRO Online will always have access to it, then it doesn't make sense if the setting is disabled and it still works.

then you would apply the force extra ram to the plugin not the game.

JoseAaronLopezGarcia, 2024-05-01:
This isn't an issue, it's normal behaviour. Games were made with the standard 24MB of RAM, forcing more can have underided results depending on the game. It should only be enabled on games that allow it.

closed 2024-05-01 by JoseAaronLopezGarcia

Dragartosaziados, 2024-05-01:
> This isn't an issue, it's normal behaviour. Games were made with the standard 24MB of RAM, forcing more can have underided results depending on the game. It should only be enabled on games that allow it.

Then that should be told on GitHub and why is the setting even enabled by default in the first place?

Dragartosaziados, 2024-05-01:
> This isn't an issue, it's normal behaviour. Games were made with the standard 24MB of RAM, forcing more can have underided results depending on the game. It should only be enabled on games that allow it.

What about [this](https://github.com/PSP-Archive/ARK-4/issues/346)?

JoseAaronLopezGarcia, 2024-05-02:
> > This isn't an issue, it's normal behaviour. Games were made with the standard 24MB of RAM, forcing more can have underided results depending on the game. It should only be enabled on games that allow it.
> 
> The repo should tell what does this setting do. What about [this](https://github.com/PSP-Archive/ARK-4/issues/346)?

I'll look into it.

Dragartosaziados, 2024-06-12:
> I'll look into it.

So, have you looked into it?