PSP-Archive/ARK-4 #362: Wrong background image in "Machi: Unmei no Kousaten Tokubetsu-hen" on PSVITA

NagaseKouichi, 2024-05-02:
### PSP/Vita Model

PCH-1000/1100 (Vita Fat)

### ARK Version

4.20.69 r4

### Homebrew/GAME (if applicable)

Machi: Unmei no Kousaten Tokubetsu-hen

### Plugins ( if applicable )

npdrm_free

### Steps to reproduce

start the game and select any character

### USB Charge

Always

### Overclock

Disabled

### Power Save

XMB

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Always

### Inferno Cache

LRU

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

On

### Hide DLC

Off

### Turn off LEDs

Disabled

### Verification of form.

- [X] I agree that I properly filled out the items listed above.

NagaseKouichi, 2024-05-02:
I don't know if it's right to submit this issue here, because adrenaline have this isuue too.

In game "Machi: Unmei no Kousaten Tokubetsu-hen" on PS Vita, the background image is wrongly dispalyed or is missing. But there is no this problem on PSP.

Here is the compare image with the same scene:

[image]

NagaseKouichi, 2024-05-02:
This issue is both in adrenaline mode and standalone mode.

JoseAaronLopezGarcia, 2024-05-02:
So it's basically an issue with the emulator? That's interesting. I'll check it out.

JoseAaronLopezGarcia, 2025-03-18:
I'm gonna need a lot more information about this. Specially since I don't speak japanese.
What does the background image say on PS Vita? Does changing CPU speeds have any effect?
Sometimes games have simple antipiracy where they check if CPU has been previously overclocked (CFW).

NagaseKouichi, 2025-03-20:
The background image says 'The image for this cut is not exsist'.
I didn't overclock PSV when I test this game. But I will test changing CPU speeds later to see if it has any effect.

JoseAaronLopezGarcia, 2025-03-24:
Any news on this? Perhaps you can play around with the new UMD delay setting?

NagaseKouichi, 2025-03-26:
> Any news on this? Perhaps you can play around with the new UMD delay setting?

I updated ARK-4 to the latest release (Rev 168), and tryed the follow settings:

change clock in game to auto,
disable memory stick speed up,
disable skip sony boot logo,

none of them has any effect. The problem is the same as before.

Where can I find the new UMD delay setting? I Didn't find that setting anywhere.

NagaseKouichi, 2025-03-26:
I tryed adding `ULJM05111, delayumd_2, on` to SETTINGS.TXT and the problem is still there.

JoseAaronLopezGarcia, 2025-03-26:
Interesting. Does this issue appear elsewhere? It could be a bug in the official emulator.

NagaseKouichi, 2025-03-27:
Yes, this is also in Adrenaline psp emu and ARK-4's standalone mode.

JoseAaronLopezGarcia, 2025-03-27:
Interesting, so it's an issue of the Official PSPemu (there are a lot of games that need fixing with pspemu). This is going to be a difficult one for me since I don't speak Japanese so I will have a hard time getting around the game.

NagaseKouichi, 2025-03-30:
It's very easy to reproduce this problem. Just select ”はじめる” -> ”はじめから” to start a new game, and select the default character 牛尾政美. Then the first scene is the scene in the screenshot.