PSP-Archive/ARK-4 #363: only loads halfway on japanese psp

MarioMasta64, 2024-05-02:
### PSP/Vita Model

PSP-3000 (Slim & Bright)

### ARK Version

4

### Steps to reproduce

1. buy a japanese psp
2. run ark cfw
3. ark cfw loads and replaces system settings info and games in iso form show up however the select menu is non-functional (the button is fine) and extras only contains 1seg comic reader and x-radar portable

im unsure about the select issue but would the different items in extras effect showing cfw settings?

### USB Charge

Disabled

### Overclock

Disabled

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Disabled

### Inferno Cache

Off

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

Off

### Hide DLC

Off

### Turn off LEDs

Disabled

### Verification of form.

- [X] I agree that I properly filled out the items listed above.

MarioMasta64, 2024-05-02:
also another note: the update option is also not replaced with ark update

nallwolf, 2024-05-02:
I have a Japanese PSP-3000 in front of me right now with the newest version of ARK-4 Live running on a freshly installed 6.61 with no problems. Either the files on your memory stick are corrupted or you potentially have a bad shoulder button (would cause the VSH menu to not show up when pressing select, but shouldn't mess with the Extras icons). You should re-download ARK and copy the savedata to your memory stick again. If you still have the problem, try testing your buttons with Key[s]Tester: https://www.gamebrew.org/wiki/KeysTester_PSP

MarioMasta64, 2024-05-02:
i redownloaded it several times
it works perfectly fine on pro-c but not ark-4
also according to the button tester every button works perfectly fine nothing is being held

MarioMasta64, 2024-05-02:
[image]
[image]
[image]
[image]
[image]

evidence of everything ive been talking about above

also its not the files cause the other psp i did right before it is fine
[image]
[image]

MarioMasta64, 2024-05-02:
perhaps it isnt specifically that its a japanese psp but in case its relevant any
[image]

MarioMasta64, 2024-05-02:
okay an update on this: for some reason the memory adapter i used was the culprit

i honestly dont know how but the comment of a corrupted memory stick helped me find this out

closed 2024-05-02 by MarioMasta64