PSP-Archive/ARK-4 #364: sstates plugin does not work properly

devratik, 2024-05-06:
### PSP/Vita Model

PSP-N1000 (PSP Go)

### ARK Version

4.20.69.r4

### Plugins ( if applicable )

ef0:/seplugins/pspstates/states_menu_pict_mod.prx 1

### Steps to reproduce

try to load a savestate and it reboots to xmb menu

### USB Charge

Always

### Overclock

Always

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Always

### Inferno Cache

LRU

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

Off

### Hide DLC

Off

### Turn off LEDs

Disabled

### Verification of form.

- [X] I agree that I properly filled out the items listed above.

krazynez, 2024-05-06:
Can you upload your PLUGINS.TXT

devratik, 2024-05-06:
[PLUGINS.TXT](https://github.com/PSP-Archive/ARK-4/files/15223176/PLUGINS.TXT)

> Can you upload your PLUGINS.TXT

here it is

krazynez, 2024-05-07:
Do you have old plugins enabled? Might need to disable that if so.

devratik, 2024-05-07:
Even after disabling it, the problem continues. First thing comes to my mind, this custom firmware uses more resource than pro-c therefore psp can't handle loading a save state. I might be wrong though.

krazynez, 2024-05-07:
I tested it yesterday on my 3K with no issues.

devratik, 2024-05-07:
Can you give me a link for the plugin? Maybe the one i use is old. (for 6.60 firmware) I am not sure though.

krazynez, 2024-05-07:
> Can you give me a link for the plugin? Maybe the one i use is old. (for 6.60 firmware) I am not sure though.

https://www.gamebrew.org/wiki/PSPStates_Menu_Pict_Mod_PSP

Aneemasta, 2024-12-20:
I can confirm the same behaviour. `PSPStates` plugin is used via [CustomHOME](https://github.com/PSP-Archive/ARK-4/wiki/Recommended-Plugins#custom-home-mod-psp), with slightly altered [conf.ini](https://github.com/user-attachments/files/18209189/conf.ini.txt) (colours, positions of elements, order of menu items) and custom background presented in `ms0:/SEPLUGINS/CUSHOME/IMAGE/ (any 480x272 24-bit BMP with ASCII-compatible name should be fine).
### Here're details:
| Param | Value |
|--------|--------|
| PSP/Vita Model | PSP-3000 (Slim & Bright) |
| ARK Version | FW 6.60 ARK 4.20.69r151 cIPL |
| Homebrew/GAME (if applicable) | [TxtoPS V](https://www.gamebrew.org/wiki/TxtoPS_PSP) |
| Plugins ( if applicable ) | [PLUGINS.TXT](https://github.com/user-attachments/files/18209089/PLUGINS.TXT) |
| USB Charge | Always |
| Overclock | Disabled |
| Power Save | Disabled |
| Balanced Energy Mode | Disabled |
| Autoboot Launcher | On |
| Disable PSP Go Pause | Off |
| Unlock Extra Memory | Always |
| Memory Stick Speedup | Always |
| Inferno Cache | LRU |
| Skip Sony Logos | On |
| Hide PIC0 and PIC1 | On |
| Hide MAC Address | On |
| Hide DLC | On |
| Turn off LEDs | Disabled |

#### Steps to reproduce:
1. Run `TxtoPS V` and wait for it to be loaded completely
2. Press `Start` to open `TxtoPS V` main menu (this will be our test non-default state we`d like to save&load)
3. Press `Home` to open `CustomHOME`
4. Select `PSPStates` and press `X`
5. Select an empty slot, press `X`, select `Yes` and press `X` again to save state
6. Wait for saving the state to be finished (when `TxtoPS V` main menu appears again)
7. In `TxtoPS V` main menu select `Reboot TxtoPS` and press `O` (the controls there are eastern-like), then press `Start` to confirm reboot
8. Wait for `TxtoPS V` to be loaded completely again
9. Press `Home` to open `CustomHOME` again
10. Select `PSPStates` and press `X` again
11. Select previously occupied slot, press `[]`, select `Yes` and press `X` again to load state

#### What was expected
- Selected savestate successfully loaded

#### What was happened
- PSP reboots completely and opens ARK `CL` instead

Aneemasta, 2025-02-22:
> I can confirm the same behaviour. `PSPStates` plugin is used via [CustomHOME](https://github.com/PSP-Archive/ARK-4/wiki/Recommended-Plugins#custom-home-mod-psp), with slightly altered [...]

Turned out my case was due to renaming some plugin files to upper-case back when #436 bug was unpatched. I've completely reinstalled cushome plugin, and the issue's gone.

JoseAaronLopezGarcia, 2025-03-29:
Closing this. Savestates work fine for me, at least when using Custom Home.

closed 2025-03-29 by JoseAaronLopezGarcia