PSP-Archive/ARK-4 #365: "Force Extra Memory" freezes the PSP

Dragartosaziados, 2024-05-07:
### PSP/Vita Model

PSP-3000 (Slim & Bright)

### ARK Version

4.20.69 Rev 4

### Homebrew/GAME (if applicable)

Spider-Man: Web of Shadows – Amazing Allies

### Steps to reproduce

If "Force Extra Memory" is set to "Always", it makes the european Spider-Man: Web of Shadows – Amazing Allies PSP ISO on PSP's ARK-4 freeze when exiting it using the PS button and the PSP to restart some seconds after.

### USB Charge

Disabled

### Overclock

Disabled

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Disabled

### Inferno Cache

Off

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

Off

### Hide DLC

Off

### Turn off LEDs

Disabled

### Verification of form.

- [X] I agree that I properly filled out the items listed above.

krazynez, 2024-05-07:
Why are you creating the same issue when this was already closed and explained?

closed 2024-05-07 by krazynez

Dragartosaziados, 2024-05-07:
> Why are you creating the same issue when this was already closed and explained?

It isn't the same issue, you told me to create issues for each one of the problems I mentioned and it wasn't fixed.

JoseAaronLopezGarcia, 2024-05-07:
> > Why are you creating the same issue when this was already closed and explained?
> 
> It isn't the same issue, you told me to create issues for each one of the problems I mentioned and it wasn't fixed.

It is the same issue. We've already explained that PSP games were NOT coded with extra RAM in mind, so enforcing it can cause some of them to break. This isn't even something we can fix in the CFW.

Dragartosaziados, 2024-05-13:
> It is the same issue. We've already explained that PSP games were NOT coded with extra RAM in mind, so enforcing it can cause some of them to break. This isn't even something we can fix in the CFW.

You're right, my mistake.

Can the "Force Extra Memory" setting break PSP games?

Also, what's the difference between Inferno Cache's "LRU" and "RR" settings?