PSP-Archive/ARK-4 #366: "ICON0 in Browser" setting freezes PS Vita's ARK-4 if "Enabled" when scrolling PSP ISOs on "Files" and makes the console lag a lot when opening ARK-4 and other apps

Dragartosaziados, 2024-05-07:
### PSP/Vita Model

PCH-2000 (Vita Slim), VTE-1000 (Vita TV)

### ARK Version

4.20.69 Rev 4

### Steps to reproduce

"ICON0 in Browser" setting freezes PS Vita's ARK-4 if "Enabled" when scrolling PSP ISOs on "Files". I can exit ARK-4 via PS button and close it, but when reopening it or opening another app, it lags a lot, so I'm forced to restart the console and set "ICON0 in Browser" to "Disabled" for it to not have both issues.

### USB Charge

Disabled

### Overclock

Disabled

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Disabled

### Inferno Cache

Off

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

Off

### Hide DLC

Off

### Turn off LEDs

Disabled

### Verification of form.

- [X] I agree that I properly filled out the items listed above.

krazynez, 2024-05-07:
Then disable it.

Dragartosaziados, 2024-05-07:
> Then disable it.

Already did, but I want the browser to show the games' icons without having these issues.

JoseAaronLopezGarcia, 2024-05-07:
> > Then disable it.
> 
> Already did, but I want the browser to show the games' icons without having these issues.

Yeah I guess it can be improved by doing async loading.

Dragartosaziados, 2024-05-07:
> > > Then disable it.
> > 
> > 
> > Already did, but I want the browser to show the games' icons without having these issues.
> 
> Yeah I guess it can be improved by doing async loading.

What do you mean?

JoseAaronLopezGarcia, 2024-05-07:
> > > > Then disable it.
> > > 
> > > 
> > > Already did, but I want the browser to show the games' icons without having these issues.
> > 
> > 
> > Yeah I guess it can be improved by doing async loading.
> 
> What do you mean?

Techhnical stuff, don't worry about it.

Dragartosaziados, 2024-06-29:
I'm not having this issue after reinstalling ARK-4.

closed 2024-06-29 by Dragartosaziados