PSP-Archive/ARK-4 #368: Latest ARK4 release crashes when loading Stage 2 on PSP 3000 6.61 OFW

MarkZ90, 2024-05-14:
### PSP/Vita Model

PSP-3000 (Slim & Bright)

### ARK Version

v1715537822

### Steps to reproduce

1. Copy ARK_01234 folder into /PSP/SAVEDATA/ folder.
2. Copy ARK_Loader folder into /PSP/GAME/ folder.
3. Run ARK Loader
4. ARK Loader hangs at:

Stage 1 Starting
Loading Stage 2 at: ms0:/PSP/SAVEDATA/ARK_01234/ARK4.BIN
Executing ARK Stage 2

5. 4.20.69 (DC10) r4 works correctly.

### USB Charge

Disabled

### Overclock

Disabled

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Disabled

### Inferno Cache

Off

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

Off

### Hide DLC

Off

### Turn off LEDs

Disabled

### Verification of form.

- [X] I agree that I properly filled out the items listed above.

MarkZ90, 2024-05-14:
It seems v1715432584 introduced the issue as the previous versions work fine.

krazynez, 2024-05-14:
We are aware thanks.

krazynez, 2024-06-12:
Should be fixed with latest release now.

krazynez, 2024-06-26:
Can you verify it works in latest release.

JoseAaronLopezGarcia, 2024-06-26:
> Can you verify it works in latest release.

Yes this was fixed.

closed 2024-06-26 by JoseAaronLopezGarcia

Varun-garg, 2024-08-10:
I got same issue in ARK-4-v1722213839-DEBUG.

renegade2k, 2025-05-21:
Got the same bug with my PSP 3004 fw 6.60 and ARK4 r180