PSP-Archive/ARK-4 #370: Problem with PSP go

Deivikk, 2024-05-20:
### PSP/Vita Model

PSP-N1000 (PSP Go)

### ARK Version

4.20.68

### Steps to reproduce

I discovered two errors on PSP go (at least on mine) Every time I force the memory to always or change the internal cache, pausing games on the PSP doesn't work, I would like to know if I did

### USB Charge

Always

### Overclock

Always

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Always

### Memory Stick Speedup

Always

### Inferno Cache

LRU

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

Off

### Hide DLC

On

### Turn off LEDs

Disabled

### Verification of form.

- [X] I agree that I properly filled out the items listed above.

JoseAaronLopezGarcia, 2024-05-20:
This is not a bug, it's normal behavior. The pause feature of the PSP Go relies on the PSP's extra 32MB, when you force that memory to go to the game instead then you leave no space for the pause feature to work, it would crash the system!
Same goes for Inferno Cache, it relies on the extra RAM and thus allowing the pause feature to work would cause a crash because they would overwrite each other's memory.

JoseAaronLopezGarcia, 2024-05-20:
On another note: Forcing Extra RAM on games is overall a very unstable experience, some games work fine, others don't like it at all. As for Inferno Cache, you might as well delete it if you have a modern high speed memory card, it's mostly useful for older/slower cards.

closed 2024-05-20 by JoseAaronLopezGarcia

Deivikk, 2024-05-20:
> On another note: Forcing Extra RAM on games is overall a very unstable experience, some games work fine, others don't like it at all. As for Inferno Cache, you might as well delete it if you have a modern high speed memory card, it's mostly useful for older/slower cards.

In my case, I use the PSP's internal memory, so it is not necessary to use the inferno cache, Right?

JoseAaronLopezGarcia, 2024-05-20:
> In my case, I use the PSP's internal memory, so it is not necessary to use the inferno cache, Right?

I'm not sure, the PSP Go's internal memory isn't as fast as today's SD cards, it might still provide some speed boost on there.

pebeto, 2024-10-21:
I consider that this must be mentioned on the project's wiki. There's a lot of misinformation on internet about this issue.

JoseAaronLopezGarcia, 2024-10-21:
> I consider that this must be mentioned on the project's wiki. There's a lot of misinformation on internet about this issue.

100% agree. There's lots of either outdated or plainly wrong information going around, that's why we made the wiki, to give users a place where they can find trustworthy information, but as you just pointed out, there's still lots we have to keep adding as we go.

@krazynez should do it.