PSP-Archive/ARK-4 #371: PSX Bilinear Filter = OFF (psp request)

Esper69, 2024-05-26:
### New Feature

I think it would be really nice and useful to have an option to disable bilinear filtering on PSX games (you can do it on PS Vita, but it's missing on PSP!).
[image]

krazynez, 2024-05-26:
That is builtin with Sony's Vita PSX emulator not sure its doable on the PSP side.

Esper69, 2024-05-26:
On the PSP, it is enabled by default, so logically it can be disabled, right? Even with some plugin, it doesn't necessarily have to be in that menu.

JoseAaronLopezGarcia, 2024-05-26:
What about setting the screen mode to original?

Esper69, 2024-05-26:
The image was just an example taken from Google. When setting custom dimensions to remove the top and bottom black bars, the image appears slightly blurry because by default, there is a bilinear filter that cannot be disabled on the PSP but can be disabled on the PS Vita.

JoseAaronLopezGarcia, 2024-05-26:
Are you sure there's a bilinear filter on the PSP? I believe it just upscales the original image.

Esper69, 2024-05-26:
Sure, in fact on the PS Vita when it is enabled, it appears as PSP, and when disabled, it is displayed as defined as in original mode but with scaled pixels without blurriness (provided you have selected a setting other than original screen, otherwise without scaling, nothing changes). Indeed, the function of the filter is precisely to smooth the pixels for a more homogeneous result, but in PSX games on PSP, it would be better to disable this option, especially on such a small screen where it is much better to have defined edges rather than smoothed ones. It looks good even without it, but since on PS Vita it can be disabled, I'm sure some developer here can manage to implement this function.

JoseAaronLopezGarcia, 2024-05-26:
The Vita can't be using the PSP's GPU to do any bilinear filtering or upscaling of any sorts since that's handled by the Vita's GPU itself as it feeds from the original PS1 VRAM.

Esper69, 2024-05-27:
ok, but in fact the ebot does not have an integrated bilinear filter so there is necessarily a way to deactivate it. maybe it would have been difficult to create it but since it already exists it can certainly be deactivated like on psvita or other emulators. deactivating this filter also on psp would be exceptional so I await a response from some programmer who understands the importance of the requested implementation.

krazynez, 2024-06-26:
> ok, but in fact the ebot does not have an integrated bilinear filter so there is necessarily a way to deactivate it. maybe it would have been difficult to create it but since it already exists it [...]

Umm @JoseAaronLopezGarcia knows the PS1 stuff VERY well.