PSP-Archive/ARK-4 #374: Real UMD delay support (for Aces of War)

Sukotto-1999, 2024-06-10:
### New Feature

Feature: 

Support for emulation of the real UMD delay. There is an anti-piracy feature confirmed by me and an other GitHub user (here: https://github.com/hrydgard/ppsspp/issues/11062) where the game stops loading on the chapter selection menu and then hangs there. 

The support for it should be added in a newer version and should be enabled by default for Aces of War (both Japanese and European versions)

And also, can a plugin that simulates real UMD delay for all ISO drivers for PRO/LME be developed?

krazynez, 2024-06-10:
We have discussed this in the past, but no moment has happened with it was more of a finding I found. Before for another issue that was brought up in the past. It is on a todo list to get done though for those few games that use it. 

As for the plugin. I am sure if we implement it someone can make a plugin with it for other CFW's.

Sukotto-1999, 2024-11-28:
You know what? Instead of waiting for the support to be added, i think i might find a way to get the protection in Aces of War removed at least if i have time.

JoseAaronLopezGarcia, 2025-03-20:
I'm gonna look into this. Implementing UMD delay is easy enough, but I'd like to have full memory stick speed instead of UMD speed. If I can't get anything working then I'll implement the UMD delay feature.

JoseAaronLopezGarcia, 2025-03-20:
So here's the thing, I've tried many many times to fix this without affecting load times, but no avail.
As soon as I implemented a UMD delay (automatic, only for this game and perhaps DJMAX) everything works fine.
So I guess for now I'll leave it like that.

One thing, so far the patch only applies to the european version, I need to know the Game ID of the Japanese version (you can get the Game ID from the ISO via the Custom Launcher, highlight the game and press triangle then go to game info).

I haven't tested it on DJMAX either but it should work.

JoseAaronLopezGarcia, 2025-03-20:
This pre-release implements UMD delay and I have tested that it solves the issues with Aces of War:
https://github.com/PSP-Archive/ARK-4/releases/tag/v1742494692

I would still need the Japanese Game ID to implement the fix there too.

JoseAaronLopezGarcia, 2025-03-20:
Found the ID for the Japanese version: ULJM-05075

I'm closing this.

closed 2025-03-20 by JoseAaronLopezGarcia