PSP-Archive/ARK-4 #377: Livearea features for ARK-X?

pmsobrado, 2024-06-16:
### PSP/Vita Model

PCH-2000 (Vita Slim)

### ARK Version

Version 4.20.69 (DC10) r8

### Steps to reproduce

I've noticed that the Livearea has some extra features for ARK-X bubbles:

[image]
[image]
[image]
[image]

They seem to be related to poketstation, however I can't see them and I haven't found any info on the wiki regarding this.

Is there a way to enable them? Is this just part of a feature that is not yet complete and will be in the future?

Thanks!

### USB Charge

Disabled

### Overclock

Disabled

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Disabled

### Inferno Cache

Off

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

Off

### Hide DLC

Off

### Turn off LEDs

Disabled

### Verification of form.

- [X] I agree that I properly filled out the items listed above.

krazynez, 2024-06-16:
Those are just generic icvons we don't have any functionality with them

pmsobrado, 2024-06-17:
Got it, thanks!

closed 2024-06-26 by krazynez

LiEnby, 2024-08-10:
these icons appear on every PS1 classic, if you *really* and are only actually enabled in a select few.

if you really want pocketstation support, you can make use of https://silica.codes/Li/PocketstationUnlocker to force unlock it on every "PS1 Classic"