PSP-Archive/ARK-4 #378: cIPL Wiki is unclear and misleading

kusazh, 2024-06-18:
### New Feature

https://github.com/PSP-Archive/ARK-4/wiki/cIPL

1. The system version should be `6.61` since the new cIPL supports only `6.61`.
2. There is no proper tutorial for using cIPL on other CFW (PRO/ME).

kusazh, 2024-06-18:
Also, I would like to translate `VSH_LANG.TXT`/`XMB_LANG.TXT` into Chinese (simplified/traditional). But I didn't find the English version of them.

JoseAaronLopezGarcia, 2024-06-18:
1. Classic cIPL is compatible with both 6.60 and 6.61, the readme states this.
2. This isn't a PRO/ME project page.

As for translations, we don't have a plain text english version, the english text is hardcoded (https://github.com/PSP-Archive/ARK-4/blob/main/extras/menus/vshmenu/trans_en.c).
For Chinese we had a user come up with a clever solution so we can use the 8 bit ascii to display special chinese characters.
More information here: https://github.com/PSP-Archive/ARK-4/pull/329

krazynez, 2024-06-18:
Number 1 is resolved. I agree with @JoseAaronLopezGarcia for number 2. If we add that users will come to us for ME and PRO support/request.

kusazh, 2024-06-18:
Thank you for clarifying.

closed 2024-06-18 by kusazh