PSP-Archive/ARK-4 #379: Recovery menu flickering on CL

maodus, 2024-06-21:
### PSP/Vita Model

PSP-3000 (Slim & Bright)

### ARK Version

FW 6.61 ARK 4.20.69 r8 cIPL

### Steps to reproduce

1. Enable recovery menu in settings
2. Open recovery menu
3. Observe flickering

I believe this is because the recovery menu is loaded as a module and tries to fight the CL's GUI thread. Could probably be fixed by hijacking `sceDisplaySetFrameBuf` if this is the case.

Heres a video of what happens:

https://github.com/PSP-Archive/ARK-4/assets/34797411/2d1ae22a-2b5f-490c-b7ba-f7692079f979

### USB Charge

Disabled

### Overclock

Always

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

On

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Always

### Inferno Cache

LRU

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

On

### Hide DLC

Off

### Turn off LEDs

Disabled

### Verification of form.

- [X] I agree that I properly filled out the items listed above.

krazynez, 2024-06-21:
Gameboot skipping is causing the issue. If disabled recovery works fine (on my end that is)

krazynez, 2024-06-22:
Fixed in [pre-release](https://github.com/PSP-Archive/ARK-4/releases/tag/v1719015930)

closed 2024-06-22 by maodus