PSP-Archive/ARK-4 #381: Request to add switch for ISO modes

Dark-Jaguar, 2024-06-25:
### New Feature

As much as I've had success with the Inferno ISO driver, an option to switch to Sony's own NP9660 would be appreciated to restore what other custom firmware solutions have.  Would this be doable?

JoseAaronLopezGarcia, 2024-06-26:
Convert the game to PBP format.

closed 2024-06-26 by JoseAaronLopezGarcia

Dark-Jaguar, 2024-06-26:
I'm not sure I understand.

krazynez, 2024-06-26:
ISO,CSO,JSO,ZSO all use Inferno2. EBOOT.PBP's use Sony NP9660

JoseAaronLopezGarcia, 2024-06-26:
Sony NP9660 is the Official Sony driver used for the Official Sony format. We don't use NP9660 for unofficial formats such as ISO.
The CFW will use one driver or the other depending on the format.

There's really not that many games that are better using Sony's format and driver.

Dark-Jaguar, 2024-11-27:
Thank you I understand now.  That said, the option was nice in other custom firmwares, and would be easier than file converting a bunch of ripped iso's, especially for quickly A/B testing.

krazynez, 2024-11-27:
> Thank you I understand now. That said, the option was nice in other custom firmwares, and would be easier than file converting a bunch of ripped iso's, especially for quickly A/B testing.

Your still not quite understanding. We don't have the other drivers at all. So if we created a option for it then it would still do nothing because the drivers would be missing.