PSP-Archive/ARK-4 #383: Randomly entering Custom Launcher when playing special movie in Kamaitachi no Yoru 2

mhwikicn, 2024-07-21:
### PSP/Vita Model

PSP-3000 (Slim & Bright)

### ARK Version

4.20.69 r9

### Homebrew/GAME (if applicable)

ULJM-05110

### Plugins ( if applicable )

CheatMaster Fusion PSP r21d

### Steps to reproduce

1. boot the game
2. choose last item in main menu
3. wait for the special movie to play
4. the playback randomly stops, then PSP will enter ARK Custom Launcher
currently disabled CheatMaster Fusion plugin for ULJM-05110 as a workaround

### USB Charge

Disabled

### Overclock

Always

### Power Save

XMB

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Always

### Memory Stick Speedup

Disabled

### Inferno Cache

Off

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

On

### Hide DLC

Off

### Turn off LEDs

Disabled

### Verification of form.

- [X] I agree that I properly filled out the items listed above.

krazynez, 2024-07-21:
Hmmm does that use LT + RT + DPAD Down + Start for some shortcut?

mhwikicn, 2024-07-21:
> Hmmm does that use LT + RT + DPAD Down + Start for some shortcut?

Yes, I know the shortcut for custom launcher, this issue happened without any key pressed

krazynez, 2024-07-21:
Interestingly enough tempAR seems to work fine on it.

JoseAaronLopezGarcia, 2025-03-16:
Any news on this?

mhwikicn, 2025-03-28:
> Any news on this?

Updated to r168, it still happens.  Is there a way to disable shortcut for launcher(LT + RT + DPAD Down + Start)?

JoseAaronLopezGarcia, 2025-04-10:
> > Any news on this?
> 
> Updated to r168, it still happens. Is there a way to disable shortcut for launcher(LT + RT + DPAD Down + Start)?

You mention CheatMaster in the report, can you elaborate what this does and why you need it?

mhwikicn, 2025-04-18:
> > > Any news on this?
> > 
> > 
> > Updated to r168, it still happens. Is there a way to disable shortcut for launcher(LT + RT + DPAD Down + Start)?
> 
> You mention CheatMaster in the report, can you elaborate what this does and why you need it?

Currently Cheat Master Fusion is not needed in this game, I'm disabling it for this particular title