PSP-Archive/ARK-4 #385: XMBIH crashes Ark-4 on PS Vita

DeadLootJ, 2024-07-29:
### PSP/Vita Model

PCH-1000/1100 (Vita Fat)

### ARK Version

4.20.69 (DC10) r11

### Plugins ( if applicable )

always, ms0:/seplugins/npdrm_free.prx, on
always, ms0:/seplugins/nploader.prx, on
xmb, ms0:/seplugins/category_lite.prx, on
xmb, ms0:/seplugins/xmbih.prx, on
xmb, ms0:/seplugins/xmbim.prx, on
xmb, ms0:/seplugins/pprefs.prx, on
game, ms0:/seplugins/TempAR/tempar.prx, on
game, ms0:/seplugins/leda.prx, off

### Steps to reproduce

Enable XMBIH in PLUGINS.txt.
Open Ark-4 in adrenaline or launch from custom bubble.
It will show that it is loading Ark-4.
About a minute or two later it will crash with a C1-2853-3 error.
This can also be achieved by installing XMBIH via Custom Launcher, upon exiting from Custom Launcher it will crash with the same error code.

### USB Charge

Disabled

### Overclock

Disabled

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Disabled

### Inferno Cache

Off

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

On

### Hide DLC

Off

### Turn off LEDs

Disabled

### Verification of form.

- [X] I agree that I properly filled out the items listed above.

JoseAaronLopezGarcia, 2024-07-29:
Which version exactly?

DeadLootJ, 2024-07-29:
1.3 fix2

JoseAaronLopezGarcia, 2024-07-29:
I just remembered that XMB IH does indeed have issues with other plugins, more info here: https://gbatemp.net/threads/release-ark-4-cfw-port-for-psp.581263/post-10070672

DeadLootJ, 2024-07-29:
I have disabled all plugins with exception of XMBIH, but it still crashes with the C1-2853-3 error. However, I don't have this problem on my PSP 1001, 3000, or Go.  As a side note, they are all on 4.20.69 (DC10) r11.  In case of pertinence, these are the plugins being used.

Go plugins:
always, ef0:/seplugins/npdrm_free.prx, on
always, ef0:/seplugins/nploader.prx, on
xmb, ef0:/seplugins/category_lite.prx, on
xmb, ef0:/seplugins/xmbim.prx, on
xmb, ef0:/seplugins/xmbih.prx, on
xmb, ef0:/seplugins/pprefs.prx, on
xmb, ef0:/seplugins/ssssspng.prx, on
game, ef0:/seplugins/TempAR/Tempar.prx, on
game, ef0:/seplugins/DS3Remapper.prx, off
game, ef0:/seplugins/cfwblock.prx, off

1001,3000 plugins:
always, ms0:/seplugins/npdrm_free.prx, on
always, ms0:/seplugins/nploader.prx, on
xmb, ms0:/seplugins/cxmb/cxmb.prx, off
xmb, ms0:/seplugins/category_lite.prx, on
xmb, ms0:/seplugins/xmbim.prx, on
xmb, ms0:/seplugins/xmbih.prx, on
xmb, ms0:/seplugins/pprefs.prx, on
xmb, ms0:/seplugins/ssssspng.prx, on
game, ms0:/seplugins/TempAR/Tempar.prx, on
game, ms0:/seplugins/SimpleBatteryViewer.prx, on

JoseAaronLopezGarcia, 2024-07-29:
ok that's a lot of great information, thanks, we will look into it as soon as we can.

DeadLootJ, 2024-07-29:
Awesome, thank you

JoseAaronLopezGarcia, 2025-03-16:
Closing as unable to reproduce the issue. Make sure you use the latest version of ARK and XMBIH.

closed 2025-03-16 by JoseAaronLopezGarcia