PSP-Archive/ARK-4 #387: PBOOT title updates aren't applied when launching games through the custom launcher, & PBOOTs overtake the detection of the original EBOOT if both are in the same game folder

Coolumin, 2024-08-02:
### PSP/Vita Model

Any

### ARK Version

4.20.69 r11

### Homebrew/GAME (if applicable)

UCUS98744, ULUS10538

### Plugins ( if applicable )

ps1, ms0:/seplugins/popsloader/popsloader.prx, on
xmb, ms0:/seplugins/categorieslite/category_lite.prx, on
game, ms0:/seplugins/dayviewer_game.prx, on
xmb, ms0:/seplugins/dayviewer.prx, on
ps1, ms0:/seplugins/popsloader/cdda_enabler.prx, on
ps1, ms0:/seplugins/popsloader/POPSCORE.PRX, on
game, ms0:/SEPLUGINS/AdhocToUSB.prx, off
game, ms0:/SEPLUGINS/SGKeyDumper.prx, off
always, ms0:/SEPLUGINS/prxshot_png_mod.prx, off
game, ms0:/SEPLUGINS/TempAR/TEMPAR.PRX, off
ps1, ms0:/SEPLUGINS/TempAR/tempar_lite.prx, off
always, ms0:/SEPLUGINS/UVC.PRX, off
xmb, ms0:/SEPLUGINS/CXMB/CXMB.PRX, off
always, ms0:/SEPLUGINS/JOYSENS.PRX, off
always, ms0:/SEPLUGINS/joysens_lite.prx, off
always, ms0:/SEPLUGINS/npdrm_free_mod.prx, on
always, ms0:/SEPLUGINS/nploader_mod.prx, on
game, ms0:/SEPLUGINS/npdecrypter.prx, off
ULUS10490, ms0:/SEPLUGINS/brightness.prx, off
always, ms0:/SEPLUGINS/brightness.prx, on
ULUS10490, ms0:/SEPLUGINS/brightness.prx, off
ULES01347, ms0:/SEPLUGINS/brightness.prx, off
game, ms0:/SEPLUGINS/Allefresher.prx, on
xmb, ms0:/SEPLUGINS/UmdImageCreator.prx, on

### Steps to reproduce

1. Download a legitimate EBOOT format game from the PlayStation Store.
2. Download the title update for the game with the update button.
3. Attempt to launch the game through the custom launcher with the setting for "Show DLC" turned off 

This can be worked around by using an .iso version of the game and update it with UMDGen or ISO_Tool. The XMB on Adrenline with ARK-4 (Strangely it doesn't seem to work with the stock version of Adrenline) or a real PSP will also natively support the PBOOT update patching. That being said, it is still unfortunate that this is a missing feature. 

It would also be nice if there was support for PBOOT updates on ISOs without the need of an external tool, just like neur0n's ME custom firmwares.

I have also mentioned this on the PSP Homebrew Discord server, but I figured I post it here too so it's more publically known. 

### USB Charge

Disabled

### Overclock

Always

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Homebrew

### Inferno Cache

Off

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

Off

### Hide DLC

On

### Turn off LEDs

Disabled

### Verification of form.

- [X] I agree that I properly filled out the items listed above.

JoseAaronLopezGarcia, 2025-03-16:
Make sure you enable "Scan DLC Entries" in Menu Settings.

JoseAaronLopezGarcia, 2025-03-16:
Should be fixed in latest (pre)release.

closed 2025-03-16 by JoseAaronLopezGarcia