PSP-Archive/ARK-4 #390: Disable PSP’s damaged supercapacitor

kusazh, 2024-08-09:
### New Feature

The damage to the PSP’s supercapacitor can cause the PSP to incorrectly recognize the battery level, so the device will refuse to boot up. I wonder if it is possible to disable the PSP’s supercapacitor via a firmware option, so that this issue can be resolved without a hardware remove / replacement.

I found some related discussions:

https://www.reddit.com/r/PSP/comments/1937flu/my_psp_3001e_was_having_some_problems_the_clock/
https://www.reddit.com/r/PSP/comments/de3yj9/just_found_a_fix_for_quick_discharge_during/
https://gbatemp.net/threads/psp-internal-battery-replacement.621721/

krazynez, 2024-08-09:
That is a hardware issue not software. You would just have to replace the capacitor if you don't want the battery to drain faster.

kusazh, 2024-08-10:
Ok, I see.

closed 2024-08-10 by kusazh