PSP-Archive/ARK-4 #392: ARK-4 Firmware: Incompatibility with .ctf Theme Plugins - Extras Category Misalignment and Launcher Inaccessibility

Plasmaphys, 2024-08-17:
### PSP/Vita Model

PSP-3000 (Slim & Bright)

### ARK Version

4.20.69 r11

### Plugins ( if applicable )

ms0:/seplugins/cxmb/cxmb.prx on

### Steps to reproduce

Incompatibility with .ctf Theme Plugins:
1. Install ARK-4 firmware on your PSP.
2. Apply any .ctf theme plugin that includes modifications to the XMB (Cross Media Bar).
3. Navigate to the Extras category where the Firmware settings and launcher are located.
4. Observe the appearance and behavior of the Extras category.
5. Note any misalignment, missing elements, or inability to access the Firmware settings.

Launcher Inaccessibility:
1. Turn on your PSP
2. Attempt to open the launcher from the Extras category.

### USB Charge

Disabled

### Overclock

Always

### Power Save

Always

### Balanced Energy Mode

Always

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Always

### Memory Stick Speedup

Always

### Inferno Cache

LRU

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

On

### Hide MAC Address

On

### Hide DLC

On

### Turn off LEDs

Disabled

### Verification of form.

- [X] I agree that I properly filled out the items listed above.

krazynez, 2024-08-17:
[image]
Working fine. Not sure what you mean. Juat launched CL also from the xmb and worked fine.

Plasmaphys, 2024-08-17:
Figured out what the problem was. It was the theme itself. it didn't have an extras menu. Which is where the firmware is located. I managed to fix both of my issues with a new theme aswell. I am sorry for reporting a false bug.

closed 2024-08-17 by Plasmaphys