PSP-Archive/ARK-4 #393: XMB only displays 111 Games in ISO folder

adelavig, 2024-08-18:
### PSP/Vita Model

PSP-2000 (Slim)

### ARK Version

ARK 4.20.69 r11

### Plugins ( if applicable )

I don't usually use plugins but I have tested with

vsh, ms0:/seplugins/categorieslite/category_lite.prx, on

### Steps to reproduce

1. Copy 134 (or more) ISO/CSOs to "ISO" folder on a 128GB Memory Stick to MicroSD adapter. (Any number above 111 shows the same issue)
2. Insert into PSP and power up
3. Select "Memory Stick" from "Game" menu, only first 111 games are displayed. Press "Square" button to categorise games and see that only 111 games are detected.
4. Remove the first 30 games from ISO folder.
5. Select "Memory Stick" from "Game" menu and see that remaining 104 games are displayed

I have tried all combinations of "Unlock Extra Memory", "MS Speedup" and "Inferno Cache" and issue persists.
I have tried using "categories_lite.prx" but issue persists. Example, put 100 games in "PSP1" folder and 34 games in "PSP2" folder.  PSP1 folder shows 100 games and PSP2 folder shows 11 games.

### USB Charge

Disabled

### Overclock

Game

### Power Save

XMB

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Always

### Memory Stick Speedup

Always

### Inferno Cache

LRU

### Skip Sony Logos

On

### Hide PIC0 and PIC1

Off

### Hide MAC Address

Off

### Hide DLC

Off

### Turn off LEDs

Disabled

### Verification of form.

- [X] I agree that I properly filled out the items listed above.

krazynez, 2024-08-18:
That is your adapter/sdcard causing that not ARK-4.

adelavig, 2024-08-18:
I'm sorry to have bothered you. You were kinda right. It wasn't the adapter, it was a corrupt game. Please feel free to send me a virtual slap.

Bug resolved. User stupidity.

closed 2024-08-18 by adelavig