PSP-Archive/ARK-4 #395: Freeze on TempAR Plugin

endyarfian, 2024-08-31:
### PSP/Vita Model

PSP-3000 (Slim & Bright)

### ARK Version

4.20

### Homebrew/GAME (if applicable)

All games that TempAR support

### Plugins ( if applicable )

game, ms0:/seplugins/TempAR/tempar.prx 1
ps1, ms0:/seplugins/TempAR/tempar_lite.prx 1

### Steps to reproduce

1. Install plugin
2. Launch game
3. launch tempAR (ps + R)
4. freeze and reboot

### USB Charge

Disabled

### Overclock

Disabled

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Disabled

### Inferno Cache

Off

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

Off

### Hide DLC

Off

### Turn off LEDs

Disabled

### Verification of form.

- [X] I agree that I properly filled out the items listed above.

krazynez, 2024-08-31:
Your plugins are missing a `,`

Sukotto-1999, 2024-10-04:
I have the same issue too, I updated it to the latest version and boom it works

closed 2025-03-16 by JoseAaronLopezGarcia