PSP-Archive/ARK-4 #397: Buttons stop respondin after opening ark loader

StolenKellogsXIII, 2024-09-05:
### PSP/Vita Model

PSP-2000 (Slim)

### ARK Version

4.20.69

### Steps to reproduce

1.install ARK as normal
2.open ARK
3.It stops all button input

### USB Charge

Disabled

### Overclock

Disabled

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Disabled

### Inferno Cache

Off

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

Off

### Hide DLC

Off

### Turn off LEDs

Disabled

### Verification of form.

- [X] I agree that I properly filled out the items listed above.

krazynez, 2024-09-06:
Make sure you don't have any plugins running incorrectly. Have a 1K-3k and a GO and all work fine.

StolenKellogsXIII, 2024-09-06:
ive fixed the problem, the select button was stuck in and was freezing the software

closed 2024-09-06 by StolenKellogsXIII