PSP-Archive/ARK-4 #400: Not working vsh/Launcher/Swttings

TheSerchCp, 2024-09-14:
### PSP/Vita Model

PSP-3000 (Slim & Bright)

### ARK Version

Ark r14 

### Steps to reproduce

1.Power on
2.Use dual sd card adapter.
3.Wait to load.
4.All previus installed plugins works.
5.Press select no showing vsh 
6.Search for ark swttings and update it doesmt apper.

When i put old mem stick vsh and swttings works, but with dual adapter none.

### USB Charge

Disabled

### Overclock

Disabled

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Always

### Memory Stick Speedup

Always

### Inferno Cache

RR

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

Off

### Hide DLC

Off

### Turn off LEDs

Disabled

### Verification of form.

- [X] I agree that I properly filled out the items listed above.

krazynez, 2024-09-14:
This is your adapters issue not ARK-4. I also have a Dual adapter and can only use 1 slot in it or it just breaks.

JoseAaronLopezGarcia, 2024-09-14:
Yeah dual slot SD adapters are known to be really bad and corrupt your SD cards. I recommend using a single slot adapter.

FULANO93, 2024-09-15:
I always use an two slots adapter on my PSP 2004 with a 32 and a 64 sd cards, for me it is working normally, only in versions 13 and 14 of ARK it occurred to me after update, when starting the PSP it did not show items in the EXTRA tab and it didn't even show the VSH menu when pressing select, but I updated it and on v15 everything is normal again

TheSerchCp, 2024-09-16:
> I always use an two slots adapter on my PSP 2004 with a 32 and a 64 sd cards, for me it is working normally, only in versions 13 and 14 of ARK it occurred to me after update, when starting the PSP [...]

For me stills doesnt works, i update to newer r15 manually, because the option isnt showing. And still sont show nothin, i use profast dual adapter with 2x64gb.

FULANO93, 2024-09-16:
The games work normally? I haven't tested this when I had a problem, but I remember that after updating, my PSP started with the producer's logos, but I had chosen the options not to show them, I had to turn the PSP off and on again a couple of times to get it back to normal and then I could update via online update... it could be that somehow the ARK save settings are not being loaded... try to save again and if nothing works, then you always have the option to reinstall the official firmware and after reinstall ARK, you can also try installing the full version.

JoseAaronLopezGarcia, 2024-10-02:
This was an issue introduced in r14, latest pre-release should have fixed it.

JoseAaronLopezGarcia, 2024-12-17:
dual sd adapters are well known to cause all sorts of issues, I'll be closing this as it's not really related to ARK.

closed 2024-12-17 by JoseAaronLopezGarcia